espressif/arduino-esp32 #109: Application crashes : CPU halted

Thoma-HauC, 2016-12-30:
Hello,

I updated the repository and compiled my pending buggy WiFi project.
It seems to work but after 20 minutes it crashes with the following output :

> Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400847f7  PS      : 0x00060a33  A0      : 0x80083dbe  A1      : 0x3ffdbf00  
A2      : 0x0000001d  A3      : 0x3ffdbf84  A4      : 0x00000004  A5      : 0x3ffd59b8  
A6      : 0x3ffdbfb0  A7      : 0x00000000  A8      : 0x800828a2  A9      : 0x3ffdbef0  
A10     : 0x00000003  A11     : 0x00060a23  A12     : 0x00060a23  A13     : 0x00000000  
A14     : 0x00000001  A15     : 0x3ffc0800  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000021  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
>
> Backtrace: 0x400847f7:0x3ffdbf00 0x40083dbe:0x3ffdbf20 0x40082b91:0x3ffdbf40 0x400e1e15:0x3ffdbf80 0x400db3d5:0x3ffdbfb0 0x400e2070:0x3ffdbfe0 0x400e237c:0x3ffdc000 0x400dac58:0x3ffdc040 0x400db07d:0x3ffdc0c0 0x400db0d5:0x3ffdc0f0 0x400d10a0:0x3ffdc110 0x4010635d:0x3ffdc130 0x400d1f18:0x3ffdc160 0x400d1f54:0x3ffdc180 0x400d0e6e:0x3ffdc1a0 0x400ff15d:0x3ffdc210
>
> CPU halted.

Could someone tell me what the origin of this issue could be?
What can I do to help to debug this issue?

If necessary, I can provide the source code of this small test application.

Thanks in advance.

Thoma

SirRibeiro, 2016-12-30:
I can confirm this, I'll get the following:

Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x40084943  PS      : 0x00060833  A0      : 0x80083f0a  A1      : 0x3ffebbb0  
A2      : 0x0000001d  A3      : 0x3ffebc34  A4      : 0x00000004  A5      : 0x3ffdd41c  
A6      : 0x3ffebc60  A7      : 0x00000000  A8      : 0x800829ee  A9      : 0x3ffebba0  
A10     : 0x00000003  A11     : 0x00060823  A12     : 0x00060823  A13     : 0x00000000  
A14     : 0x3ffc0840  A15     : 0x00000010  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000021  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x40084943:0x3ffebbb0 0x40083f0a:0x3ffebbd0 0x40082cdd:0x3ffebbf0 0x400e1cd1:0x3ffebc30 0x400db291:0x3ffebc60 0x400e1f2c:0x3ffebc90 0x400e22d0:0x3ffebcb0 0x400dace9:0x3ffebcf0 0x400daead:0x3ffebd20 0x400d1052:0x3ffebd40 0x401062d8:0x3ffebd60 0x400d0dc1:0x3ffebd80 0x400d0e3c:0x3ffebde0 0x400ff05d:0x3ffebe20

CPU halted.

I was thinking it could be the alimentation of the esp32, but it appends with the usb and with direct power to the processor. So it must be something else...

igrr, 2016-12-30:
Backtrace should have all the info — you can throw those 0x4xxxxxx addresses info `xtensa-esp32-elf-addr2line` or into `xtensa-esp32-elf-gdb` `disass` command. Or just upload the .elf file which corresponds to this log.

me-no-dev, 2016-12-30:
I actually left it unanswered to remind me to enable esp32 in the exception decoder plugin :)

Thoma-HauC, 2016-12-30:
Hi Igrr,

`xtensa-esp32-elf-addr2line -e sketch_dec15a.ino.elf 0x400847f7`
returns
`/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c:214`

So it seems the exception thrown in FreeRTOS source.

Thoma

igrr, 2016-12-30:
list.c panicked because it was given an invalid pointer (see EXCVADDR). It would be useful to go a few stack frames back to see where the call originated from. It's likely an access to a member of an object which has a null pointer (EXCVADDR = 0x00000021).

me-no-dev, 2016-12-30:
Not neceserely :) there should be more addresses that could be decoded to give better trace. I hope to upload the plugin later today.

Thoma-HauC, 2016-12-30:
Igrr,

Here is the whole backtrace information :
`/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c:214
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c:4398
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1998
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/freertos/sys_arch.c:500
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/tcpip.c:467
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/api_lib.c:811
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/api_lib.c:811
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/sockets.c:3233
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/sockets.c:3233
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/sockets.c:3233
/home/xxxxxxx/sketchbook/hardware/espressif/esp32/cores/esp32/Print.h:50
/home/xxxxxxx/sketchbook/hardware/espressif/esp32/libraries/WiFi/src/WiFiClient.cpp:158
/home/xxxxxxx/sketchbook/hardware/espressif/esp32/cores/esp32/Stream.cpp:279
/home/xxxxxxx/sketchbook/hardware/espressif/esp32/cores/esp32/Stream.cpp:287
/home/xxxxxxx/sketchbook/sketch_dec15a/sketch_dec15a.ino:133
/home/xxxxxxx/sketchbook/hardware/espressif/esp32/cores/esp32/main.cpp:11 (discriminator 1)

Thoma

me-no-dev, 2017-01-01:
@Thoma-HauC the trace above is incomplete because it's missing function names. There are two ways to overcome this:
- Install [ESPExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder) and use that
- run addr2line with the following arguments ```xtensa-esp32-elf-addr2line -aipsfC -e [elf]``` to get the values missing

me-no-dev, 2017-01-01:
it looks like you are trying to read from possibly closed and freed connection. Could we have a look at some part of the code? Does it crash immediately or does it run a few times before crashing?

Thoma-HauC, 2017-01-01:
Hi,

Here is the code : [source-code.txt](https://github.com/espressif/arduino-esp32/files/680080/source-code.txt)
I just remove some unnecessary comments and debug messages.

As said initially, it crashes after a randomly time.

This time I got the below exception :
>Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400e8704  PS      : 0x00060430  A0      : 0x800975c4  A1      : 0x3ffda6e0  
A2      : 0x3ffdcce8  A3      : 0x3ffc1740  A4      : 0xffffffe2  A5      : 0x3ffd3838  
A6      : 0x00000000  A7      : 0x000000e8  A8      : 0x800eebe2  A9      : 0x00000001  
A10     : 0x3ffc1740  A11     : 0x3ffdcce8  A12     : 0x00000000  A13     : 0x0000003a  
A14     : 0x3ffd59d0  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xffffffe2  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
>
>Backtrace: 0x400e8704:0x3ffda6e0 0x400975c4:0x3ffda770
>
>CPU halted.

and
>xtensa-esp32-elf-addr2line -aipsfC -e sketch_jan01a.ino.elf 0x400e8704:0x3ffda6e0 0x400975c4:0x3ffda770

returns
>0x400e8704: ieee80211_ifattach at ??:?
0x400975c4: ppTask at ??:?

But the other exception thrown too:
>Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400847f7  PS      : 0x00060a33  A0      : 0x80083dbe  A1      : 0x3ffdbfc0  
A2      : 0x0000001d  A3      : 0x3ffdc044  A4      : 0x00000004  A5      : 0x3ffd5a04  
A6      : 0x3ffdc070  A7      : 0x00000000  A8      : 0x800828a2  A9      : 0x3ffdbfb0  
A10     : 0x00000003  A11     : 0x00060a23  A12     : 0x00060a23  A13     : 0x00000000  
A14     : 0x00000001  A15     : 0x3ffc081c  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000021  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
>
>Backtrace: 0x400847f7:0x3ffdbfc0 0x40083dbe:0x3ffdbfe0 0x40082b91:0x3ffdc000 0x400e1dad:0x3ffdc040 0x400db36d:0x3ffdc070 0x400e2008:0x3ffdc0a0 0x400e2314:0x3ffdc0c0 0x400dabf0:0x3ffdc100 0x400db015:0x3ffdc180 0x400db06d:0x3ffdc1b0 0x400d1050:0x3ffdc1d0 0x401062f5:0x3ffdc1f0 0x400d1eb0:0x3ffdc220 0x400d1eec:0x3ffdc240 0x400d0e56:0x3ffdc260 0x400ff0f5:0x3ffdc2d0
>
>CPU halted.

and in this case
>xtensa-esp32-elf-addr2line -aipsfC -e sketch_jan01a.ino.elf 0x400847f7:0x3ffdbfc0 0x40083dbe:0x3ffdbfe0 0x40082b91:0x3ffdc000 0x400e1dad:0x3ffdc040 0x400db36d:0x3ffdc070 0x400e2008:0x3ffdc0a0 0x400e2314:0x3ffdc0c0 0x400dabf0:0x3ffdc100 0x400db015:0x3ffdc180 0x400db06d:0x3ffdc1b0 0x400d1050:0x3ffdc1d0 0x401062f5:0x3ffdc1f0 0x400d1eb0:0x3ffdc220 0x400d1eec:0x3ffdc240 0x400d0e56:0x3ffdc260 0x400ff0f5:0x3ffdc2d0

returns
>0x400847f7: uxListRemove at list.c:214
0x40083dbe: xPortGetCoreID at tasks.c:4398
 (inlined by) xTaskRemoveFromEventList at tasks.c:3018
0x40082b91: xQueueGenericSend at queue.c:1998
0x400e1dad: sys_mbox_post at sys_arch.c:500
0x400db36d: tcpip_send_api_msg at tcpip.c:467
0x400e2008: tcpip_apimsg at api_lib.c:811
0x400e2314: netconn_recved at api_lib.c:811
0x400dabf0: lwip_recvfrom at sockets.c:3233
0x400db015: lwip_recvfrom_r at sockets.c:3233
0x400db06d: lwip_recv_r at sockets.c:3233
0x400d1050: WiFiClient::read(unsigned char*, unsigned int) at Print.h:50
0x401062f5: WiFiClient::read() at WiFiClient.cpp:158
0x400d1eb0: Stream::timedRead() at Stream.cpp:279
0x400d1eec: Stream::readStringUntil(char) at Stream.cpp:287
0x400d0e56: loop at sketch_jan01a.ino:116
0x400ff0f5: loopTask(void*) at main.cpp:11 (discriminator 1)

which is similar to the previous backtrace.

Thoma

henrik1235, 2017-01-02:
Got that crash too. Code was working fine before on ESP8266 and ESP32. After resetting the ESP32 the crash no longer occurs.

```
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x40084c67  PS      : 0x00060033  A0      : 0x8008422e  A1      : 0x3ffdc930
A2      : 0x0000001d  A3      : 0x3ffdc9b4  A4      : 0x00000004  A5      : 0x3ffd6380
A6      : 0x3ffdc9e0  A7      : 0x00000000  A8      : 0x80082d12  A9      : 0x3ffdc920
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060023  A13     : 0x00000000
A14     : 0x3ffdcb00  A15     : 0x3ffc10ec  SAR     : 0x00000008  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000021  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
```

Stacktrace:
```
0x40084c67: uxListRemove at list.c:214
0x4008422e: xPortGetCoreID at tasks.c:4398
 (inlined by) xTaskRemoveFromEventList at tasks.c:3018
0x40083001: xQueueGenericSend at queue.c:1998
0x400e8c81: sys_mbox_post at sys_arch.c:500
0x400e2341: tcpip_send_api_msg at tcpip.c:467
0x400e8edc: tcpip_apimsg at api_lib.c:811
0x400e91e5: netconn_send at api_lib.c:811
0x400e1c63: lwip_sendto at sockets.c:3233
0x400e1ef1: lwip_sendto_r at sockets.c:3233
0x400d2137: WiFiUDP::endPacket() at Print.h:50
<some of my code>
```

tcflanag, 2017-01-02:
This looks and sounds pretty similar to an issue I'm debugging as well, except mine is coming from a UDP server.  

````
0x400847f7: uxListRemove at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 214
0x40083dbe: xPortGetCoreID at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4398
:  (inlined by) xTaskRemoveFromEventList at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 3018
0x40082b91: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1998
0x400e2905: sys_mbox_trypost at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/freertos/sys_arch.c line 500
0x400dbef4: tcpip_inpkt at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/tcpip.c line 467
0x400dbf2c: tcpip_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/tcpip.c line 467
0x40106adb: wlanif_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/netif/wlanif.c line 184
0x400f8135: tcpip_adapter_sta_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/tcpip_adapter/./tcpip_adapter_lwip.c line 658
0x400f371a: sta_input at ?? line ?
0x400f3f5c: sta_rx_cb at ?? line ?
0x40095d4c: ppRxPkt at ?? line ?
0x4009760e: ppTask at ?? line ?
````
My barebone server can fail in 10-20 minutes if I'm really hammering the data, but slower data rates just make it take longer (though not proportionally)
```` 
  #include <WiFi.h>
  #include <WiFiUDP.h>
  
  const char * networkName = "fake";
  const char * networkPswd = "fake";
    
  const int LED_PIN = 5;
   
  WiFiUDP Udp; 
   
  void setup() {
    pinMode(LED_PIN, OUTPUT);
    WiFi.begin(networkName, networkPswd);   
    Udp.begin(0x1936);
  }

void loop() {
  char packetBuffer[2000];
  int noBytes = Udp.parsePacket();
  if ( noBytes ) {
    Udp.read(packetBuffer,noBytes); // read the packet into the buffer
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
  }
}
````

JeroenVanOort, 2017-01-03:
I was trying to sync my ESP32 to an NTP server last night and I was experiencing some crashes too.
The script would print the current time to serial every second and sync to NTP using a modified version of https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/NTPClient/NTPClient.ino every ten seconds.

I began and closed the udp connection before and after each sync, and it crashed while trying to begin a connection after the first or second successful sync. After removing the close command and beginning the connection in setup(), it would crash 4 or 5 seconds after the second sync. Also, some of the text I was outputting to serial to debug would be garbled, as if something was messing up memory.

I've just started working with the ESP32 a few days ago, so I don't know too much about it but I'd love to help test things where possible.

Thoma-HauC, 2017-01-09:
I was thinking about me-no-dev's comment : _you are trying to read from possibly closed and freed connection_

Does this mean that between client.available() and client.readStringUntil(), the connection was closed ?
If yes, what in the below code has closed the connection ?

`  WiFiClient client;`
`while (client.available())`
`  {`
`    String line = client.readStringUntil('\r');`
`    if (line.length() == 1)`
`    {`
`      break;`
`    }`
`  }`

Thoma

me-no-dev, 2017-01-09:
@Thoma-HauC you can read up to available() even after the client has closed the connection (data might still be in the buffer). in your case you might not read all of the data or line.length() will return 0.

Looking again at the traces above and having played with udp for some time (work stuff). It seems that the problem comes from the netconn api somewhere, because for work I used the raw api and have been really pushing the limits and have not had a single crash. The netconn/socket API is the way to go though, so it would be awesome if we can pinpoint the problem and fix it in idf.

JimWatV1, 2017-01-11:
I am getting this, too, using the SparkFun ESP32.  I took the example code here https://learn.sparkfun.com/tutorials/esp32-thing-hookup-guide#arduino-example-wifi that hits example.com and changed the loop to hit it every second instead of when the button is pressed.  It will crash after a minute or so with Error of type LoadProhibited.

tcflanag, 2017-01-11:
Could someone else who is also seeing crashes try adding this to your code:

````
void loopTask(void *pvParameters)
{
    setup();
    for(;;) {
        loop();
    }
}

extern "C" void app_main()
{
    initArduino();
    xTaskCreatePinnedToCore(loopTask, "loopTask", 4096, NULL, 1, NULL, 0);
}
````

It swaps your code to run on Core0 instead of Core1.  It seems to clean up the crashes for me, though I still can't figure out why.

me-no-dev, 2017-01-11:
yeah, but that leads to using only one of the cores for everything, which is not what we want.

JorenSix, 2017-01-11:
I run in the issue as well using a sparkfun ESP32 Thing. After about five minutes of sending packages over UDP (at about 50Hz) the error looks similar:
```
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x40084807  PS      : 0x00060433  A0      : 0x80083dce  A1      : 0x3ffdc280  
A2      : 0x0000001d  A3      : 0x3ffdc304  A4      : 0x00000004  A5      : 0x3ffd5c4c  
A6      : 0x3ffdc330  A7      : 0x00000000  A8      : 0x800828b2  A9      : 0x3ffdc270  
A10     : 0x00000003  A11     : 0x00060423  A12     : 0x00060423  A13     : 0x00000000  
A14     : 0x3ffdc450  A15     : 0x3ffc09b0  SAR     : 0x00000012  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000021  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x40084807:0x3ffdc280 0x40083dce:0x3ffdc2a0 0x40082ba1:0x3ffdc2c0 0x400e3629:0x3ffdc300 0x400dcce9:0x3ffdc330 0x400e3884:0x3ffdc360 0x400e3b8d:0x3ffdc380 0x400dc60b:0x3ffdc3c0 0x400dc899:0x3ffdc420 0x400d1ce7:0x3ffdc450 0x400d1031:0x3ffdc480 0x40100dc5:0x3ffdc4e0

CPU halted.
```

tcflanag, 2017-01-11:
@me-no-dev, sure I'll agree that it's a lousy hack, but it's better than crashing.  I'm just wondering if it helps anyone else's crashes.

JorenSix, 2017-01-11:
I can confirm that the work-around helps. It made my application stable (without any noticeable negative side effects). Thanks for the response

Thoma-HauC, 2017-01-12:
I also confirm that the problem has not reappeared the last 12 hours with this change. What conclusions can we draw ?

Thoma-HauC, 2017-01-17:
My program has been running on both cores for 2 hours.
A modification in last commit (35316de) probably solve the issue.
Could someone else confirm ?

I will start a long time test.

tcflanag, 2017-01-18:
Nope git latest still crashes for me on my stress test.  Interestingly EVCADDR has changed though.

````
 Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled
 Register dump:
 PC      : 0x40084f1b  PS      : 0x00060633  A0      : 0x800843f8  A1      : 0x3ffdab00
 A2      : 0x0000001d  A3      : 0x3ffdab84  A4      : 0x00000004  A5      : 0x3ffcb77c
 A6      : 0x3ffdd5a0  A7      : 0x00000000  A8      : 0x80082c64  A9      : 0x3ffdaaf0
 A10     : 0x00000003  A11     : 0x00060623  A12     : 0x00060623  A13     : 0x00000018
 A14     : 0x3ffc3c28  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
 EXCVADDR: 0x0000002d  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

0x40084f1b: uxListRemove at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 212
0x400843f8: xPortGetCoreID at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4410
:  (inlined by) xTaskRemoveFromEventList at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 3030
0x40083007: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 2040
0x400e3c9d: sys_mbox_trypost at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/freertos/sys_arch.c line 562
0x400db792: tcpip_inpkt at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/tcpip.c line 473
0x400db7cc: tcpip_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/tcpip.c line 473
0x4010b4cf: wlanif_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/port/netif/wlanif.c line 171
0x400fc87d: tcpip_adapter_sta_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/tcpip_adapter/./tcpip_adapter_lwip.c line 663
0x400f70b2: sta_input at ?? line ?
0x400f7968: sta_rx_cb at ?? line ?
0x400967f8: ppRxPkt at ?? line ?
0x4009b6fd: wdevProcessRxSucDataAll at ?? line ?
0x40098232: ppTask at ?? line ?

tcflanag, 2017-01-18:
Here's my stress test: https://github.com/tcflanag/jubilant-eureka. It causes a halt in <20 minutes. Often as fast as 5.  Don't plan on doing much on your wifi network while it's running though.

igrr, 2017-01-19:
FYI, this issue is tracked in ESP-IDF here: https://github.com/espressif/esp-idf/issues/242

Thoma-HauC, 2017-01-21:
Hi,

I would like to perform instrumentation of the code of 'esp-idf' but how to rebuild it to use it with 'arduino-esp32' ?

Thoma

me-no-dev, 2017-01-22:
@Thoma-HauC what do you mean by "rebuild it to use it with 'arduino-esp32'"? How to run arduino-esp32 with IDF is in the README of this project, but I guess you are actually asking something else?

Thoma-HauC, 2017-01-22:
@me-no-dev I will try to explain with an example.
If I make a modification in _components/lwip/api/api_lib.c_, what is the full compilation process to see the effect of the modification with the Arduino IDE ?

me-no-dev, 2017-01-22:
@Thoma-HauC there is no point to see it in Arduino IDE :) if you find problem there, it should go to https://github.com/espressif/esp-idf and not here. Also to run your sketch while building all from source, you can follow the instructions to first install esp-idf and then https://github.com/espressif/arduino-esp32#using-as-esp-idf-component

me-no-dev, 2017-02-22:
Hey guys, test now!

JimWatV1, 2017-02-23:
Great work!  It's been running about 30 minutes straight, and done over 1700 HTTP GETs without any problems.  Thank you so much.

Alexandravq, 2017-03-02:
Gracias por la ayuda, yo tuve el mismo problema, me arrojaba el error CPU halted, lo que hice fue cambiar la fuente de alimentación por una de mayor corriente y funciono perfectamente. Espero que les funcione.

Thoma-HauC, 2017-03-05:
Hello me-no-dev,

From my point of view the crash does not longer happen.
Many thanks for the solution.

Thoma

closed 2017-03-05 by Thoma-HauC

burjiscooper, 2017-05-20:
New to Arduino platform here. So after me-no-dev said "hey guys try it now". What does that mean everyone did to try the fix? And how can I see what me-no-dev changed to verify my library files/code match and are up to date?

MrJustPaul, 2017-07-18:
Like burjiscooper, I am new to Arduino, getting this same error (pops right away), but I have no idea how to "try it now"!  I'd LOVE to!  Can you go slow?  Thanks in advance for patience - but I don't even know what questions to ask - just anxious to see some of the magic this little ESP32 can do!