espressif/arduino-esp32 #110: UDP based NTP Time Client example does not work on esp32 after the first time through loop()

palomarcoder, 2016-12-30:
I have the NTP time client from https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/NTPClient/NTPClient.ino working on an esp8266.  I modified it slightly to run on an esp32 by changing ESP8266Wifi.h to Wifi.h A packet is only returned the first time through the loop.  After that, I always get "no packet yet".  I have tried increasing the delay(1000) to 10 seconds, 20 seconds, and get the same behavior.  I added a debug print to the loop (this did not change the behavior) to verify that the IP lookup is working correctly each time.  

modified loop:

  WiFi.hostByName(ntpServerName, timeServerIP); 

  Serial.print("Time Server IP: ");   /* added */
  Serial.println(timeServerIP);      /* added */

Output:
WiFi connected
IP address: 
10.0.0.31
Starting UDP
Time Server IP: 129.6.15.30
sending NTP packet...
packet received, length=48
Seconds since Jan 1 1900 = 3692108698
Unix time = 1483119898
The UTC time is 17:44:58
Time Server IP: 129.6.15.30
sending NTP packet...
no packet yet
Time Server IP: 129.6.15.30
sending NTP packet...
no packet yet
Time Server IP: 131.107.13.100
sending NTP packet...
no packet yet
Time Server IP: 132.163.4.103
sending NTP packet...
no packet yet
Time Server IP: 216.228.192.69
sending NTP packet...
no packet yet
Time Server IP: 216.228.192.69
sending NTP packet...
no packet yet

palomarcoder, 2016-12-30:
I also had to comment out udp.localPort() because that function appears to not be implemented.

palomarcoder, 2017-01-01:
A small update:  It appears that if I move the WifiUDP constructor into the loop, and make a new instance every time, it works almost every time.  If I call udp.stop() ,  I immediately get a CPU Halt.  There appear to be still some issues in the udp implementation.

me-no-dev, 2017-01-02:
thanks for the updates. I'll have a look at what might not be closing the socket and get it fixed :)

tcflanag, 2017-01-02:
I've been calling Udp.flush(); at the bottom of my loop.  It works similarly to rerunning Udp.begin(), but seems slightly more stable.  Based on the way the code is written now, that seems to be the only way to get a second packet.

As far as I can see the rx_buffer delete block never gets called:
````c
int WiFiUDP::read(){
  if(!rx_buffer) return -1;
  int out = rx_buffer->read();
  if(!rx_buffer->size()){
    cbuf *b = rx_buffer;
    rx_buffer = 0;
    delete b;
  }
  return out;
}
````
As far as I can see, the cbuf::size function isn't being used properly in the UDP server. It is assuming the size changes as data is read out, but looking at cbuf.cpp, the size shouldn't change unless you explicitly resize it.

me-no-dev, 2017-01-02:
Thanks for noting this! Please pull the latest git and give it another go

me-no-dev, 2017-01-02:
yeah I must have been looking at older version of cbuf while writing UDP. ```size``` used to have the functionality of ```available```

palomarcoder, 2017-01-02:
That did the trick!  Thanks @me-no-dev  and @tcflanag !

closed 2017-01-02 by palomarcoder