espressif/arduino-esp32 #111: ESP8266WebServer does not work due to missing Fs.h

palomarcoder, 2016-12-31:
I'm trying to get the WebServer support to work.  I copied the ESP8266WebServer library folder to the Arduino\hardware\espressif\esp32\libraries folder, and modified include in the .h file to be Wifi.h rather than ESP8266Wifi.h.  Now it fails with missing Fs.h.  I assume this is the SPIFFS filesystem.  Is that implemented yet on esp32, and if so where can I find it?

forthlightning, 2017-01-20:
Any updates? Thanks for all the hard work!

MarcFinns, 2017-02-08:
Hi there, any news on this front? thanks all!

me-no-dev, 2017-02-08:
Most of the components needed to get the server running are written and in testing or waiting merges upstream in IDF. When everything needed is available and working, I will add/port the server over.

Curclamas, 2017-02-22:
How is the status regarding the webserver? Having the library would enable us to use/port the really useful lib WiFiManager (https://github.com/kentaylor/WiFiManager). 
They have already an open issue regarding this topic:
https://github.com/tzapu/WiFiManager/issues/241

me-no-dev, 2017-02-22:
still the same, waiting for the file system requirements to be merged in IDF which will happen really soon now. Network should be fine.

copercini, 2017-02-27:
@Curclamas I made a small example similar to WifiManager using the available libraries of ESP32 

It's pretty simple (and ugly) portal, but works: https://github.com/copercini/esp32-iot-examples/blob/master/WiFi_portal/WiFi_portal.ino

me-no-dev, 2017-03-10:
FS is implemented, SPIFFS will go next, currently can be used with SD and SDMMC

closed 2017-03-10 by me-no-dev

Curclamas, 2017-03-14:
@me-no-dev : Is it possible to have the ESP8266WebServer in the arduino-esp32 repo by default? Or do we have ro resort on copying the ESP8266 version an fix it manually? It sounds like all dependencies are met now. Maybe we should also come up with a more general name like "WebServer" or something to avoid confusion.