espressif/arduino-esp32 #112: Touch pins

ionciubotaru, 2016-12-31:
Please add some examples howto use the touch capability.
Do we still need the external capacitor and resistor, or everything is inside the capsule ?

tolson2000, 2017-01-13:
This example works on GPIO4 which is Touch0.
```
// ESP32 Touch Test
// teo2017012.01
// Just test touch pin - Touch0 is T0 which is on GPIO 4.

void setup() {
  Serial.begin(115200);
  delay(1000); // give me time to bring up serial monitor
  printf("\n ESP32 Touch Test\n");

}

void loop() {

  // Touch0, T0, is on GPIO4
  Serial.println(touchRead(T0));  // get value using T0

  Serial.println(touchRead(4));   // get value using GPIO 4

  Serial.println("");
  delay(1000);
}
```

tolson2000, 2017-01-13:
Add interrupt handler.

```
// ESP32 Touch Interrupt Test
// teo2017012.01
// Just test touch pin - Touch0 is T0 which is on GPIO 4.
// teo20170112.02
// Add interrupt handler

int threshold = 40;

void setup() {
  Serial.begin(115200);
  delay(1000); // give me time to bring up serial monitor
  printf("\n ESP32 Touch Interrupt Test\n");
  touchAttachInterrupt(T0, gotTouch, threshold);
}

void loop() {

  // Touch0, T0, is on GPIO4
  Serial.println(touchRead(T0));  // get value using T0

  Serial.println(touchRead(4));   // get value using GPIO 4

  Serial.println("");
  delay(1000);
}

void gotTouch(){
  Serial.println("Touched\n");
}

```

DiegoMendoza1, 2017-01-29:
Hi I have adapted the example to make two interrupts with two touch pins, at first it works quite well but after a while doing the interruptions the serial is blocked and does not show the loop or interruptions. Is it a code problem?
I show you the code I've "done". An adaptation of @tolson2000 ´s code

```
//ESP32 Touch Interrupt Test
// teo2017012.01
// Just test touch pin - Touch2 is T2 which is on GPIO 2 - Touch3 is T3 which is on GPIO 15.
// teo20170112.02
// Add interrupt handler

int threshold = 40;

  void setup() {
  Serial.begin(115200);
  delay(1000); // give me time to bring up serial monitor
  printf("\n ESP32 Touch Interrupt Test\n");
  touchAttachInterrupt(T2, gotTouch, threshold);
  touchAttachInterrupt(T3, gotTouch1, threshold);

}

void loop() {

  Serial.println(touchRead(T2));  // get value using  T2

  Serial.println(touchRead(T3));   // get value using T3

  Serial.println("");
  delay(1000);
}

 void gotTouch(){
 Serial.println("Touched1\n");
}

void gotTouch1(){
 Serial.println("Touched2\n");
}

```

me-no-dev, 2017-01-29:
first do not read the touch pins in the loop. Then instead of that, set in the interrupts a boolean flag and read it in the loop. something like this:
```cpp
int threshold = 40;
bool touch1detected = false;
bool touch2detected = false;

void gotTouch(){
 touch1detected = true;
}

void gotTouch1(){
 touch2detected = true;
}

void setup() {
  Serial.begin(115200);
  delay(1000); // give me time to bring up serial monitor
  printf("\n ESP32 Touch Interrupt Test\n");
  touchAttachInterrupt(T2, gotTouch, threshold);
  touchAttachInterrupt(T3, gotTouch1, threshold);
}

void loop(){
  if(touch1detected){
    touch1detected = false;
    Serial.println("Touch 1 detected");
  }
  if(touch2detected){
    touch2detected = false;
    Serial.println("Touch 2 detected");
  }
}
```

tolson2000, 2017-01-29:
Right, really shouldn't do any printing within interrupt handlers. Get in and out as quickly as possible.
If you need the value read it in the handler. But do any calculations or printing, if needed, in the loop.

DiegoMendoza1, 2017-01-29:
Thanks, now it works perfectly

closed 2017-02-05 by ionciubotaru

palmerabollo, 2017-04-30:
@tolson2000 I'd like to know the reason why you shouldn't do any processing in the interrupt handlers. Do they block something? Any link about this subject would be helpful.

ionciubotaru, 2017-04-30:
You cannot use delay() inside the interrupt or any other function which use timers, because inside interrupt the clock is not working. A good programming practice is to keep the function as short as possible.

juliusbangert, 2017-08-03:
What reason would there be for T1 (GPIO0) not detecting any touches at all while the other do?

me-no-dev, 2017-08-04:
it has a pullup in order for the board to boot

JurgenG, 2019-08-13:
While you can't use a delay inside interrupts, you can use millis() to use something similar to a debounce function.

JurgenG, 2019-08-13:
Does anyone have an example of touchAttachInterrupt responding differently to a long touch (e.g. > 2 seconds) versus a short touch?

babylonx, 2019-10-03:
It is true that timing functions stop updating inside interrupts but you can still get their current value. For example you can use millis() but you cannot except that it changes while in the interrupt function. For de-bouncing I do the following:

`
volatile unsigned long sinceLastTouch = 0;
void touchcall() {
	if (millis() - sinceLastTouch < 500) return;
	sinceLastTouch = millis();
	<rest of your interrupt code here>
}`

Queequack, 2020-04-14:
Your code is not working properly. For some reason, T2 didn't work on me. I chose T0 and T3. I put together the code from me-no-dev and babylonx. Every time you touch one pin and then the other, an interrupt is executed for both (not only for the one that has been touched!).

Try this code. Take into account the 500ms waiting time after each touch. Always touch pin T0 and pin T3 alternately. "Value" should always be incremented and decremented. In fact the result always remains the same, because a wrong interrupt routine is triggered when changing to the other pin (incremented and decremented at once).

Why?

`int threshold = 60;
bool touchdetected0 = false;
bool touchdetected3 = false;

int value = 0;
volatile unsigned long sinceLastTouch0 = 0;
volatile unsigned long sinceLastTouch3 = 0;

void gotTouch0(){
 if (millis() - sinceLastTouch0 < 500) return;
 sinceLastTouch0 = millis();
 touchdetected0 = true;
}

void gotTouch3(){
 if (millis() - sinceLastTouch3 < 500) return;
 sinceLastTouch3 = millis();
 touchdetected3 = true;
}

void setup() {
  Serial.begin(115200);
  delay(1000); // give me time to bring up serial monitor
  printf("\n ESP32 Touch Interrupt Test\n");
  touchAttachInterrupt(T0, gotTouch0, threshold); // T0 = GPIO 4
  touchAttachInterrupt(T3, gotTouch3, threshold); // T3 = GPIO 15
}

void loop(){
  if(touchdetected0){
    touchdetected0 = false;
    Serial.print("Touch 0 detected - Value: "); Serial.println(value=value-1);
  }
  if(touchdetected3){
    touchdetected3 = false;
    Serial.print("Touch 3 detected - Value: "); Serial.println(value=value+1);
  }
}`

Queequack, 2020-04-15:
Hi, with a little help I found the solution. As compact as the code in the interrupt function is, it is not compact enough. When an interrupt occurs, nothing should happen except setting a flag.
The correct way is to compare the time since the last touch with a delay value within the loop() function. 

In the code below the interrupt is executed on every touch (flag is set to true) This flag is queried in the loop() function. If it is true, it is first set to false again (the touch was pressed was finally detected). The time of the "successful" touch input is stored for the next comparison: sinceLastTouchT0 = millis();.
Only then does a swapped-out function query whether enough time has passed since the last touch: if (touchDelayComp(sinceLastTouchT0)). If the function returns false, the if-query is finished and nothing happens. However, if it returns true, the value "Value" is modified in this case.

```
#include <Arduino.h>

const uint8_t threshold = 60;
bool touchedT0 = false; // Touch 0 ist GPIO4
bool touchedT3 = false; // Touch 0 ist GPI15

const long touchDelay = 350; //ms
int value = 0;

volatile unsigned long sinceLastTouchT0 = 0;
volatile unsigned long sinceLastTouchT3 = 0;

bool touchDelayComp(unsigned long);

void IRAM_ATTR T0wasActivated() { touchedT0 = true; }
void IRAM_ATTR T3wasActivated() { touchedT3 = true; }

void setup() {
  Serial.begin(115200);
  delay(1000);
  touchAttachInterrupt(T0, T0wasActivated, threshold);
  touchAttachInterrupt(T3, T3wasActivated, threshold);
  Serial.println("Ready...");
}

void loop() 
{
  if (touchedT0) 
  {
    touchedT0 = false;
    sinceLastTouchT0 = millis();
    if (touchDelayComp(sinceLastTouchT0))
    {
      value = value - 1;
      Serial.print("T0: "); Serial.println(value);
    }
  }

  if (touchedT3) 
  {
    touchedT3 = false;
    sinceLastTouchT3 = millis();
    if (touchDelayComp(sinceLastTouchT3))
    {
      value = value + 1;
      Serial.print("T3: "); Serial.println(value);
    }
  }
}

bool touchDelayComp(unsigned long lastTouch)
{
  if (millis() - lastTouch < touchDelay) return false;
  return true;
}
```