espressif/arduino-esp32 #113: Remote Peripheral Integration

Xplorer001, 2017-01-02:
Is there a plan to integrate ESP32 built-in Remote Peripheral?  I am working on a Open Universal Remote. Was able to use the ESP-IDF examples with minor tweaks to test it on actual IR NEC Remote hardware.  Wanted to know if you've planned for Arduino Library and high level APIs for the same, so that I could work on those lines and contribute?

igrr, 2017-01-02:
If you are interested in contributing something like an IRRemote library port, that would be great!

me-no-dev, 2017-01-02:
I hope to have time for it in a few days. It's next in line for integration :)

igrr, 2017-01-02:
Well, RMT is a fairly simple peripheral... I just needed to use it to decode PPM signals from an RC transmitter, and ended up working with registers directly rather than with the ESP-IDF library. So I don't know how much of a HAL we need in Arduino. Probably less than what IDF provides.

me-no-dev, 2017-01-02:
As it is with most drivers so far :)

Xplorer001, 2017-01-02:
I will follow the IRRemote library structure like below, and port it. It can be later integrated with the same lib for ESP32. 

Any suggestions for better function names, structure are welcome :)

```
IRremote.h
	--header for IR send and Recevice functions. 

irSend.cpp 
irSend Class 
   |-- enableIRout() //For configure remote channel, assigned pin, IR frequency 	
   |
   |-- sendRaw(): Sends raw IR data from a buffer passed to it.
   |
   |-- mark()
   |
   |-- space()
   |
   |-- 
   

irRecv.cpp
irRecv Class
   |
   |-- enableIRIn() // configure and enable specific IR channel
   | 
   |-- decode() //calls protocol specific decode functions
   |
   |-- other functions as needed. 
   
  	

IRremote.cpp
   |
   |--  ISR()// to handle receive data and fill the buffer

IRremoteInt.h
   |
   |-- generic header  and defines for specific protocols.

ir_NEC.cpp, ir_RC5.cpp etc., will contain protocal specific functions
   |
   |--IRsend()
   |
   |--IRrecv()
`
```

kriswiner, 2017-01-02:
Please include examples when you commit to the Arduino core; I would like to learn how to use this as well. Thanks!

Xplorer001, 2017-01-07:
@kriswiner sure!
@me-no-dev  Is rtc.h and rtc.c (pre-compiled) not updated to the latest IDF ?  The library code I'm working works with the rtc.c from the latest version of ESP-IDF. So I'm including it locally in the library folder and continuing with the development. . . .

me-no-dev, 2017-01-07:
@Xplorer001 IDF libs were updated a day ago, so you should be fine

copercini, 2017-08-19:
@Xplorer001 Any update about it?

Xplorer001, 2017-08-24:
@copercini check the PR my colleague  have created on the IRRemmote https://github.com/z3t0/Arduino-IRremote/pull/491 This need to be worked on a little more. However we doing a altogether different implementation using the builtin remote peripheral. I will post a link here once it is tested more.

Vorms, 2018-01-11:
Hello, 
I am very intersted for your work !
Definitively we need a library using RMT compatible with the IRRemote library.
Thanks a lot in advance !
Just a question, I see the IR send pin is by default on pin 5 of the ESP32. Is thatpossible to change that even UART pin ? Many thanks for your help.
Thierry

Darryl-Scott, 2018-04-17:
Hi all
I spend a bit of time going through the RMT library and many examples out there. 
I have created a library that might help people.
It basically gets the RAW IR receiver data and then resends it out via another pin to an IR LED. 
Example library link: https://github.com/Darryl-Scott/ESP32-RMT-Library-IR-code-RAW

Thanks
Darryl

malinoy, 2019-04-05:
https://github.com/Darryl-Scott/ESP32-RMT-Library-IR-code-RAW

I found that this is the only working library for ESP32 DevModule. However, it sends only the RAW code that it receives via the IR receiver. Is anyone able to edit or modify this library to send an IR code (eg NEC code when the switch is pressed).

lbernstone, 2019-04-05:
https://github.com/lbernstone/IR32

malinoy, 2019-04-08:
I thank you for linking to the library.
Sorry, no examples work.

1. RMTLib_Demo: "Error compiling for board ESP32 Dev Module"
2. Loopback: "LG_timing was not declared in this scope"

lbernstone, 2019-04-09:
Ok, try again.  Changed the function params, but not the example.

malinoy, 2019-04-10:
Thank you very much for your help and effort. Now it can compile (Loopback) and load on a board , but unfortunately, they do not send any code via the IR diode. I went through other outputs pins, on GPI01 and GPI03, it sent a code on these two outputs, but each time the code is different and is not sent at the interval specified in the program, but randomly.

lbernstone, 2019-04-10:
Don't use gpio 1&3.  Those are in use by Serial.

malinoy, 2019-04-10:
I also used all the other pins, but unfortunately I could not send an IR code.
Before I used ESP8266 board and the IRsend.h library  and I was able to send an IR codo easily. On the ESP32, however, no library seems to work the same way.

closed 2019-08-15 by stale[bot]