espressif/arduino-esp32 #116: Wire.h as esp-idf component

SirRibeiro, 2017-01-04:
Hello,

It's possible to use the wire.h library using arduino-esp32 as a esp-idf component?

This is the error i get on make:

`In file included from C:/esp32_projects/arduino/components/arduino/cores/esp32/Print.h:27:0,
                 from C:/esp32_projects/arduino/components/arduino/cores/esp32/Stream.h:26,
                 from C:/esp32_projects/arduino/components/arduino/libraries/Wire/src/Wire.h:30,
                 from C:/esp32_projects/arduino/main/main.c:10:
C:/esp32_projects/arduino/components/arduino/cores/esp32/Printable.h:25:1: error: unknown type name 'class'
 class Print;
 ^
C:/esp32_projects/arduino/components/arduino/cores/esp32/Printable.h:33:1: error: unknown type name 'class'
 class Printable
 ^
C:/esp32_projects/arduino/components/arduino/cores/esp32/Printable.h:34:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^
In file included from C:/esp32_projects/arduino/components/arduino/cores/esp32/Stream.h:26:0,
                 from C:/esp32_projects/arduino/components/arduino/libraries/Wire/src/Wire.h:30,
                 from C:/esp32_projects/arduino/main/main.c:10:
C:/esp32_projects/arduino/components/arduino/cores/esp32/Print.h:34:1: error: unknown type name 'class'
 class Print
 ^
C:/esp32_projects/arduino/components/arduino/cores/esp32/Print.h:35:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^
In file included from C:/esp32_projects/arduino/components/arduino/libraries/Wire/src/Wire.h:30:0,
                 from C:/esp32_projects/arduino/main/main.c:10:
C:/esp32_projects/arduino/components/arduino/cores/esp32/Stream.h:38:1: error: unknown type name 'class'
 class Stream: public Print
 ^
C:/esp32_projects/arduino/components/arduino/cores/esp32/Stream.h:38:13: error: expected '=', ',', ';', 'asm' or '__attribute__' before ':' token
 class Stream: public Print
             ^
In file included from C:/esp32_projects/arduino/main/main.c:10:0:
C:/esp32_projects/arduino/components/arduino/libraries/Wire/src/Wire.h:34:1: error: unknown type name 'class'
 class TwoWire: public Stream
 ^
C:/esp32_projects/arduino/components/arduino/libraries/Wire/src/Wire.h:34:14: error: expected '=', ',', ';', 'asm' or '__attribute__' before ':' token
 class TwoWire: public Stream
              ^
C:/esp32_projects/arduino/components/arduino/libraries/Wire/src/Wire.h:97:1: error: unknown type name 'TwoWire'
 extern TwoWire Wire;
 ^
make[1]: *** [/c/esp-idf/make/component_wrapper.mk:177: main.o] Error 1
make: *** [C:/esp-idf/make/project.mk:377: main-build] Error 2
`

My code:

`#include "freertos/FreeRTOS.h"
#include "esp_wifi.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"
#include "driver/gpio.h"

#include "Arduino.h"
#include <Wire.h>

esp_err_t event_handler(void *ctx, system_event_t *event)
{
    return ESP_OK;
}

void app_main(void)
{
    nvs_flash_init();
    tcpip_adapter_init();
    initArduino();
	
    ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
    ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    wifi_config_t sta_config = {
        .sta = {
            .ssid = "fluxodata",
            .password = "fluxodata01",
            .bssid_set = false
        }
    };
    ESP_ERROR_CHECK( esp_wifi_set_config(WIFI_IF_STA, &sta_config) );
    ESP_ERROR_CHECK( esp_wifi_start() );
    ESP_ERROR_CHECK( esp_wifi_connect() );
	
	
    pinMode(13, OUTPUT);
    
    int level = 0;
    while (true) {
        digitalWrite(13, level);
        level = !level;
		printf("%d\n",level);
        vTaskDelay(500 / portTICK_PERIOD_MS);
    }
	
	
}

`

me-no-dev, 2017-01-04:
you are running this from C file and Arduino is C++ :) change to cpp and things will work (see readme)

SirRibeiro, 2017-01-04:
I totally miss that.

Thank you very much

SirRibeiro, 2017-01-04:
@me-no-dev do you know how to use external libraries?
Putting them in \arduino\components\arduino\libraries does not seem to work...

forthlightning, 2017-01-04:
The IDF doesnt look in your regular library folder. 

If you check out the docs for build environment, you can add your library folder as a include directory. Gotta edit the project makefile!

SirRibeiro, 2017-01-04:
Thanks for the info :)

closed 2017-01-05 by me-no-dev