espressif/arduino-esp32 #118: I2C Slave implementation

Curclamas, 2017-01-05:
# Description
It seams that the I2C slave functionality from the Arduino Wire Lib is not implemented.

# Code example
```
#include <Wire.h> 

void setup() 
{ 
 Wire.begin(0x03);
 Wire.onRequest(requestEvent);
} 

void loop() 
{ 
 delay(10); 
} 

void requestEvent() 
{
 Wire.send(0x13);
}
```

# Error occured during compilation
```
exit status 1
'class SDClass' has no member named 'truncate'
```

gryzli133, 2017-07-24:
did you find a solution for that issue?

lettierimarco, 2018-05-14:
no news? i've done with esp-idf but arduino way is simpler and clearer

lbernstone, 2018-05-14:
@stickbreaker is working on slave I2C (https://github.com/stickbreaker/arduino-esp32/tree/master/libraries/Wire). Not sure where he is at with it, but development is active.

lettierimarco, 2018-05-14:
Found out  it...
many errors in compile...

Nightreaver, 2019-05-27:
Updates?

stickbreaker, 2019-05-27:
I've got a working Slave mode, but I am still working through coexistence with Master mode.  Slave mode on ESP32 has to call `onRequest()` before the Slave is addressed. So the outbound data is always stale. Currently Slave mode exists in a separate task, and I haven't decide on the 'correct' design to refresh this data: 
* never 
* every time `onReceive()` activates
* manually controlled by application
* every time a Master transactions supplant Slave mode. 
* so some combination of all of these.

It is base on a re-implementation of `Wire()` that is designed for multi-tasking support Multiple `Wire()` objects are create, each Object is designed as single threaded, The multiple `Wire()` Objects are safely multi-threaded and share the hardware.  But, the Application must transactionalize their access if necessary.  The transaction locking with MUTEX's can be abused to cause deadlocks.  The 'simple' Arduino 'it just work' philosophy is tough to implement with complex interactions.

Chuck.

Zhairgling, 2019-06-27:
Hi @stickbreaker , Any news ?
Can you share code ? 
Thanks

canDry, 2019-07-25:
BUMP!  Anyone have a slave receive solution/fix/workaround?

nockieboy, 2019-07-28:
BUMP! Surely there must be enough demand for this feature to warrant some serious development from those that know how?

dpharris, 2019-07-28:
Update?  How can we help?

JAICHANGPARK, 2019-08-14:
Any Update This issues? 
I'd like to use I2C Slave mode with arduino uno

ameer1234567890, 2019-08-14:
@JAICHANGPARK i2c slave does work on the Arduino Uno. This issue is for ESP32.

JAICHANGPARK, 2019-08-14:
@ameer1234567890  I already know.
I mean that I hope to use Esp 32 Slave Mode using arduino IDE

yeckel, 2019-11-06:
Any updates? Or example how to workaround that issue?

Tom5051, 2019-11-21:
Please help, I also need to use an ESP32 in slave mode but get this error: 
(.literal._Z5setupv+0x8): undefined reference to `TwoWire::onReceive(void (*)(int))'

stickbreaker, 2019-11-23:
@Tom5051 @yeckel No updates yet, I have a proof of concept, but it is not ready for prime time, (It requires a major change of existing code). Possibly Spring 2020 I might have code is releasable.

Chuck.

netkruzer, 2020-02-01:
Thanks for your efforts Chuck. Another bump for i2c as a slave mode when you have a chance. Would be really useful.

stickbreaker, 2020-02-01:
@netkruzer Not ready yet. It's not yet code that I want my name attached to.  And, I do not want to even think of supporting it as is. 

Chuck.

ellensp, 2020-03-16:
Darn, I'm also wanting this feature..

zamorae, 2020-04-21:
@stickbreaker Hi Chuck. Any date in the foreseeable future that you can provide to have a release of the I2C solution for ESP32? I have been dreaming for a while to use it for some projects I have. Let us know!

Thanks
-zamorae

ameer1234567890, 2020-05-29:
It was reported on #1697 by @laercionit that "Someone had succeeded in putting as SLAVE?" Any details on this?

laercionit, 2020-05-29:
I couldn't ... I don't think TwoWire is ready for that.

gutierrezps, 2020-06-17:
I have made a workaround library based on the only two methods available from ESP-IDF.

https://github.com/gutierrezps/ESP32_I2C_Slave

sgrd, 2020-07-04:
@gutierrezps
How to use your library in ESP32(slave) communication with RasPi(master) ?
May you give an example of a compatible Python script ?

gutierrezps, 2020-07-07:
@sgrd I'm sorry, but I don't work with Raspberry Pi's at the moment, so I'm afraid I can't help you. You can look into `WireSlaveRequest` and try to map Arduino functions into the Raspberry's functions.

MkLHX, 2020-09-17:
3-4 after this issue is opened, nothing change?

Maybe need more people ask for this feature...

Morbidbystandard, 2020-11-27:
It would be nice to have an arduino master and an Esp32 slave to get I²C communication into home assistant.

gutierrezps, 2021-02-21:
@Morbidbystandard check my library: https://github.com/gutierrezps/ESP32_I2C_Slave

sgrd, 2021-02-22:
> @Morbidbystandard check my library: https://github.com/gutierrezps/ESP32_I2C_Slave
Excellent ! Thank you for your creation

EddyManay, 2021-03-24:
Hola @gutierrezps, gracias por crear una excelente librería, me funciona excelente, pero. ¿ Cuando quiero enviar un dato tipo flotante no funciona correctamente, se puede enviar datos de ese tipo ?

38saibot, 2021-03-24:
Still no Arduino support after all the years?
Hard to believe that such a basic thing like I2C communication between two ESPs is such a big problem.
With Arduino it is [So simple](https://www.arduino.cc/en/Tutorial/LibraryExamples/MasterReader) 

The ESP_I2C_Slave library mentioned above is pretty cool, but too slow and produces errors if I want to read some bytes from the slave every 5ms.

ESP32 as Master and Arduino Mega as Slave is working very good and fast, but I really do want to use 2 ESPs for that

ianr34, 2021-04-04:
the problem is the underlying esp32 i2c slave model - it's ok for getting data to the slave (task with a loop and a wait state), but hopeless for sending data to the master. Until expressif fix it 

Given how fast the esp32 is I've thought of just doing a software one anyway...I'm not sure why the above one is slow, as I can do software i2c on a 328P and do transactions in under 500uS..

John-WL, 2021-04-19:
I think a software version of this feature could help a lot while this is worked on. 

Does anyone know a library/repo I can look into for that matter? In my case, 30 minutes of google search yielded nothing but slave libraries for other microcontrollers and master libraries.

ianr34, 2021-04-21:
I've been keeping am eye on the new esp chips coming out, and it still seems to be a problem. As a result I'm still using 328p's even though they cost about the same as a esp32, as they are easy to make a i2c slave.. 

Looking at the tech reference manual there doesn't even appear to be the interrupts to write something with the inbuilt hardware to do a salve properly ie the slave needs an interrupt when the master is requesting data - and have the ability IN that interrupt for providing the data to be sent back. The Slave also needs an interrupt when a complete packet has arrived from the master.
Until expressif provide the ability to do that, their chips are effectively useless as i2c slaves - so I don't think you will be seeing anything in the ardunio-esp32 any time soon!

Quite a few people have used them as a spi slave instead, as that seems to be a better developed bit of hardware, but the problem I have is pin shortages and board size - and the i2c is already broken out and on the board...

samhoff20, 2021-06-16:
4.5 years later and there's still no officially supported solution???

38saibot, 2021-06-17:
> 4.5 years later and there's still no officially supported solution???

To say it in Metallica's words: Sad but true

Tom5051, 2021-06-17:
It is unresolved issues like this, that make the ESP32 a totally worthless and useless chip for real world usage. I have stopped using it in my projects because I find I spent more time designing / coding around unfinished or non working issues. It is total joke that there are outstanding issues like basic stuff - I2C comm not working, non linear ADAC, memory leaks, etc etc etc etc etc etc

ianr34, 2021-06-17:
The i2c master works fine - not the ardunio one of it (last I checked it was still a bit flacky) but if you do the direct esp32 calls it is rock solid.
I also haven't had memory leaks - that could be that I've got out of the ardunio libraries as well.

It's the i2c slave that is the real problem - the esp32 just needs a interrupt or function callback when a request is received, so you can fill in the data and send it back. Simple and obvious. 
Yet still not there (and not the ardunio library fault) - people have written all sorts of workarounds, which is nuts. At some point I'll have to go and write a decent software slave for it - which is also nuts. Until then, I use other things to be the slave devices, even the esp32 is cheaper and can do more!

me-no-dev, 2021-06-18:
it is very easy to say that something is very "basic" yet not working. The reality is very different than that. We have tried to make I2C slave work in a few occasions but have failed to be able to fit the hardware into the software. If you get to know the hardware more, you would end up at the same situation. Original I2C slave was written for AVRs that have a very different I2C peripheral than what the ESP32 chips have. You can get something going if you use the IDF API, so you do have that option, it's not like you are left in the dirt. Unfortunately at this point we are not yet able to imagine I2C slave into the Arduino API

Tom5051, 2021-06-18:
Well I build an automatic dog door for a greyhound using bluetooth tags. Unfortunately the ESP32 was the only board I could find that was remotely able to detect the tags. If it wasn't for that, I would NEVER use an ESP32 ever again. It was a nightmare because I had to spend to much time researching work arounds for things that just work with Arduino boards. I'm glad you guys enjoy stuffing around trying to butcher this thing to work properly but for the average user, too much effort for bugger all gain. I wouldn't even rate the ESP32 series as a development board because it hasn't even passed BETA stage. Should never have been released.

ianr34, 2021-06-18:
> it is very easy to say that something is very "basic" yet not working. The reality is very different than that. We have tried to make I2C slave work in a few occasions but have failed to be able to fit the hardware into the software.

me-no-dev - I understand the silicon is the problem, but I was really hoping that later chips than the esp32 had fixed it - ie the s2 etc. But, nope. And other cheap chips can do it (10c sensor chips can!), so it is really something that needs to be fixed!

As for doing it with the idf - which I do use - can you point to any example that works like we would all expect ie **both** a interrupt/ callback/queue to wait on when a i2c packet arrives (with the full packet ready to be accessed) and a interrupt or callback when the master requests data - so that it can be dynamically generated (quickly) and sent back.

If you can do that with the direct idf I haven't seen it (and I regularly search). And looking at the doc on the expresif site it doesn't seem to be doable either.  Putting data in a buffer that will be read by the master at some unspecified time in the future simply isn't workable (I can't think of a use case of that at all). 

By the way, if you could show me an example of how to do it with the idf (a proper client) it wouldn't take take long to do one for the ardunio libraries.. 

So, in summary,we **are** left in the dirt as far as I can see. I'm designing a board now ( prob a few thousand will be made) and I can't use the esp32 for the i2c slave controlling a number of sensors. Which is real annoying. I could fix it by doing a software i2c client, but don't have the time to write that and make in solid..

It's a really big hole in the whole esp32 lineup..

thetinycat, 2021-06-22:
I just started working with the ESP32 and the first thing I needed to do was to implement one as an I2C-slave. So... this is going great. Can someone recommend another small devboard that supports bluetooth and is usable as I2C-slave?

ferrveciana, 2021-07-12:
This sucks. :(

ferrveciana, 2021-07-12:
@gutierrezps is there a way I can read using your library, with slave_reciever on my esp32, and sending the messages from an Arduino nano 33 BLE that uses the standard library? Or can your library be used from an Arduino nano in any way?
Thanks for your work

Tom5051, 2021-07-12:
I used Serial1 for comms. Just have to keep checking the serial1 buffer for new commands / messages

ianr34, 2021-07-12:
there is NO need to poll if you use the serial port- it is properly set up by expresif so you can either get an interrupt when something comes in, or wait on the buffer queue (which is a rtos queue). 
It's a pity the expresif team doing the serial didn't talk to the team doing i2c and tell them what they needed to do..

me-no-dev, 2021-10-11:
If you are still interested, I2C Slave is on the way: https://github.com/espressif/arduino-esp32/pull/5746

JAICHANGPARK, 2021-10-11:
@me-no-dev awesome

ianr34, 2021-10-11:
OK, so you are saying that the esp32 won't be supporting the full i2c slave model - but the s2 onwards will. That's grreat, as I could move to the s2 for the slave, and will prob be going to the s3 generally anyway (more pins!) - price willing.

Of course, I could just write a bit banging slave for the esp32 - looking at the above code late last night I was assuming that was what you had half done, otherwise you can't support **Wire.onRequest(requestEvent);**  - as it of course generates the bytes to be returned to the master..

So is the above code only for the -s2 and onwards?

me-no-dev, 2021-10-12:
it work on all chips, but on ESP32 it just behaves a bit differently. You need to pre-load the data that the master will be able to read. ESP32 as master is also not liking having the clock stretched for too long. Let's say you use S2 as Slave and ESP32 as Master. The slave needs to handle the onRequest as fast as possible to ensure that the ESP32 will not give up waiting for the clock to stop stretching.

ianr34, 2021-10-12:
yes, so it doesn't work on the esp32. 

Preloading the data IS the core issue and has been since the esp32 was released - preloading is no good for many many many things. No point the master getting data that might be very old..

And yes,  onRequest (or whatever it is called as I don't use the ardunio wire library) is always done fast. Typically you just copy the data from somewhere safe, and then exit. You never do any work in it...

closed 2021-10-14 by me-no-dev

btperez, 2022-04-10:
@me-no-dev I'm not sure if I have a full understanding from the comments, but how can I go about implementing an ESP32 as an I2C slave in Arduino or has the problem not been fixed?