espressif/arduino-esp32 #120: analogRead seems to speed up system clock

JeroenVanOort, 2017-01-08:
I'm building an energy monitor using [EmonLib](https://github.com/openenergymonitor/EmonLib) and I've got most things working now. Since I'll need to add a time stamp to the measurements I send to my web application, I'd like to sync the ESP32's system clock to an NTP server. Sync to NTP doesn't seem to be my problem so far, but keeping time is. I've observed that the system clock on my device seems to go faster when doing a lot of analogRead's, which is necessary for monitoring energy usage.

After a measurement that takes a second, I print the time to serial. I've seen 3 to 4 seconds jumps and sometimes it's just the one second like I'd expect it to. When I comment out the measurement function and put `delay(1000);` in it's place, time runs normal.

Do you have any clue where this behavior comes from and how we could fix this?

JeroenVanOort, 2017-01-13:
I've made a simple program to test this. I observe that after 30 seconds according to my laptop's clock, about 50 seconds have passed according to the ESP32's clock. When I comment out the for loop, the ESP32's clock runs just fine.

What can we do to fix this?

```c
#include <time.h>

time_t lcd_time;

void setup()
{
    Serial.begin(115200);
}

void loop()
{
    int i;
    for (i = 0; i < 100000; i++) {
      analogRead(36);
    }
    
    time(&lcd_time);
    Serial.println(lcd_time);
    
    delay(1000);
}
```

igrr, 2017-01-13:
Thanks for the example. I don't have a workaround for you, need to investigate this. The issue is in the esp-idf, not in the Arduino libraries.

igrr, 2017-01-16:
We were able to reproduce this, and this looks like a hardware bug, i.e. a write to `FRC_TIMER_INT_REG(0)` gets lost, so the timer interrupt doesn't get cleared... and for some reason this happens when ADC is in use. We will try to reproduce this in the simulation environment.

For now, the workaround is to add a write to some random memory address into the `frc_timer_isr` function (components/newlib/time.c:85), for example
```
WRITE_PERI_REG(DR_REG_FRC_TIMER_BASE + 0x60, 0xabababab);
```

It can go before or after `WRITE_PERI_REG(FRC_TIMER_INT_REG(0), FRC_TIMER_INT_CLR);` (both ways seem to fix the problem).

We'll update ESP-IDF with this workaround while we are trying to narrow down the root cause.

tolson2000, 2017-01-17:
This workaround does seem to work using the esp-idf with arduino as component.
I don't know how to move this to the arduino-esp32 though to use with Arduino IDE.

igrr, 2017-01-17:
The fix is staged for merging into the ESP-IDF. Once it is merged, @me-no-dev can update libraries in this repository to make the fix available in Arduino IDE.

closed 2017-01-17 by igrr

g01d10x, 2018-03-25:
Hello ESP32 community :)

Does anyone know if this was ever addressed by @me-no-dev on the Arduino side of things? I'm doing some ADC related work and noticed that time seems to accelerate for the ESP32 vs the rest of the world.