espressif/arduino-esp32 #121: std::list fails to link

sticilface, 2017-01-09:
see error
```
/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_469322/sketch/sketch_jan09b.ino.cpp.o:(.literal.setup+0x4): undefined reference to `std::__detail::_List_node_base::_M_hook(std::__detail::_List_node_base*)'
/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_469322/sketch/sketch_jan09b.ino.cpp.o: In function `setup':
/Users/amelvin/git/arduino-esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/4.8.5/bits/stl_list.h:114: undefined reference to `std::__detail::_List_node_base::_M_hook(std::__detail::_List_node_base*)'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Dev Module.
```

code example
```
#include <list>

class abc {
public:

private:
};

std::list<abc*> _list;

void setup() {
  abc * coolptr; 
  _list.push_back( coolptr ) ; 
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

jkearneyma, 2017-01-10:
I found that adding ' -lstdc++' at the end of 'compiler.c.elf.libs=' in platform.txt fixed this.

sticilface, 2017-01-10:
and indeed that did fix it!  cheers. i'll leave it open as an issue to fix

me-no-dev, 2017-01-20:
that is merged now so closing :)

closed 2017-01-20 by me-no-dev