espressif/arduino-esp32 #122: touchRead

DiegoMendoza1, 2017-01-11:
Hi, I am looking for information about the capacitive touch and as they are programmed in the ide of arduino, I have seen that it is necessary to read them with the function touchRead but I do not find which is the librery that needs to recognize this function.

can anybody help me?

Thank you

me-no-dev, 2017-01-11:
no library is needed ;) just call ```touchRead(pin);``` on any pin that supports touch.

DiegoMendoza1, 2017-01-11:
I have done so, but when I build with the arduino ide it says touchRead is not declared

me-no-dev, 2017-01-11:
You must have selected the wrong board or something. touchRead is defined [here](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-touch.h#L43);

DiegoMendoza1, 2017-01-11:
The board is correct, I have the esp devkit. I have written this code as an example and still does not recognize that touchRead is declared:

`#include <arduino.h>

int Touch = 20;

void setup() {

Serial.begin(115200);

pinMode(Touch, INPUT);

}

void loop() {
  
touchRead(Touch);

 if(Touch == LOW){

  Serial.println("ENABLE");
  
  delay(1000);

  
 else{
  Serial.println ("DISABLE");
  
  delay(1000);
 }
  
  }
}`

me-no-dev, 2017-01-11:
this sketch compiled just fine here. Also "Arduino" is capitalized. Also touchRead will return unsigned integer value, not LOW/HIGH.

me-no-dev, 2017-01-11:
btw are you sure you are running the latest esp32-arduino?

DiegoMendoza1, 2017-01-11:
I´m not sure, where is it?

DiegoMendoza1, 2017-01-11:
I found it, I didn´t have the latest version, now it compiles correctly.

Thanks

closed 2017-01-11 by DiegoMendoza1