espressif/arduino-esp32 #123: HardwareSerial Flow Control

forthlightning, 2017-01-11:
Is this something that is currently or will be enabled?

Looking specifically at [this usage in esp-idf example.](https://github.com/espressif/esp-idf/blob/master/examples/20_uart/main/uart_test.c#L156)

I had ben using the [uart api here](https://github.com/espressif/esp-idf/blob/master/components/driver/uart.c), but found it tricky to work with, so switched to the [uart-hal functions here.](https://github.com/espressif/arduino-esp32/blob/56cd8f63f0454adc8b770c486c460ca95c16ba1c/cores/esp32/esp32-hal-uart.h) but my sensor is lookin' to control the flow and I see no option to allow it to do its thing.

forthlightning, 2017-01-11:
Additional question same topic:

What is the significance of [these](https://github.com/espressif/arduino-esp32/blob/f5d6828c5a2924a1c41a6bdb70fb65e8aeaa20ae/cores/esp32/esp32-hal-uart.h#L29) definitions in the uart-hal?

I see SERIAL_8N1 listed as a default in the HardwareSerial.h library [here](https://github.com/espressif/arduino-esp32/blob/56cd8f63f0454adc8b770c486c460ca95c16ba1c/cores/esp32/HardwareSerial.h#L40) so that's what I used, not sure why. Are the others listed other configuration options?

me-no-dev, 2017-01-16:
these definitions are for bits, stop bits and paring :)
I have not written any hardware flow support in the driver mainly because Arduino has no such thing, so libraries would not expect it. Can you think of a test case that I can make to get this tested and added?

forthlightning, 2017-01-31:
This is solved! For reference here is how I did it. I used a combination of GPIO api and arduino delaymicros. I'm sure there is an esp-idf handle for microsecond delays but arduinos work for now. 

```
            // flow control for MAX485 modbus to serial transceiver
            gpio_set_level( GPIO_NUM_5, HIGH );
            delayMicroseconds( 500 );  // trial and error setting for proper sensor response
            sdm_serial.write( sdmarr, FRAMESIZE - 1 );
            delayMicroseconds( 8300 ); // frame transmission length plus some wiggle room
            gpio_set_level( GPIO_NUM_5, LOW );
            sdm_serial.flush(); // don't continue until all serial data has been sent, for stability
```

closed 2017-01-31 by forthlightning

forthlightning, 2017-06-26:
for reference this is what i was looking for:

https://github.com/espressif/esp-idf/pull/667

mattncsu, 2019-06-11:
> I have not written any hardware flow support in the driver mainly because Arduino has no such thing, so libraries would not expect it. Can you think of a test case that I can make to get this tested and added?

Perhaps communicating with LTE modems at high speeds (up to 4Mbps for the SIM7000 ([UART design note](https://simcom.ee/documents/SIM7000x/SIM7000%20Series%20UART%20Application%20Note_V1.00.pdf)) might benefit from hardware flow control)

> Flow control is very important during the transmission (large data) between the module (DCE) and the terminal device (DTE).  When receiving buffer reaches its capacity, the receiving device should be capable of pause the sending device until it overflows. SIM7000 module is  designed as no flow control by default, but users can enable this function by AT command.