espressif/arduino-esp32 #124: smart config support

seopyoon, 2017-01-12:
So far, the arduino sdk is not supporting the SmartConfig function it seems.
When are you planning on adding WiFi.beginSmartConfig()? 

Is there a way to use smart config functionality at the very moment?

seopyoon, 2017-01-13:
I made modifications to WiFiSTA.cpp and .h to support ESP's smart config functionality.  But it does not seem to work.

To WiFiSTA.cpp, I added:
```

bool WiFiSTAClass::_smartConfigStarted = false;
bool WiFiSTAClass::_smartConfigDone = false;

bool WiFiSTAClass::beginSmartConfig() {
    if (_smartConfigStarted) {
        return false;
    }

    if (!WiFi.enableSTA(true)) {
        return false;
    }

    esp_err_t err;
    err = esp_smartconfig_start(reinterpret_cast<sc_callback_t>(&WiFiSTAClass::_smartConfigCallback), 1);
    if (err == ESP_OK) {
        _smartConfigStarted = true;
        _smartConfigDone = false;
        return true;
    }
    return false;
}

bool WiFiSTAClass::stopSmartConfig() {
    if (!_smartConfigStarted) {
        return true;
    }

    if (esp_smartconfig_stop() == ESP_OK) {
        _smartConfigStarted = false;
        return true;
    }

    return false;
}

bool WiFiSTAClass::smartConfigDone() {
    if (!_smartConfigStarted) {
        return false;
    }

    return _smartConfigDone;
}

void WiFiSTAClass::_smartConfigCallback(uint32_t st, void* result) {
    smartconfig_status_t status = (smartconfig_status_t) st;
    if (status == SC_STATUS_LINK) {
        wifi_sta_config_t *sta_conf = reinterpret_cast<wifi_sta_config_t *>(result);

        esp_wifi_set_config(WIFI_IF_AP, (wifi_config_t *)sta_conf);
        esp_wifi_disconnect();
        esp_wifi_connect();
    } else if (status == SC_STATUS_LINK_OVER) {
        WiFi.stopSmartConfig();
    }
}
```

and, to WiFiSta.h, I added:
```
public:
    bool beginSmartConfig();
    bool stopSmartConfig();
    bool smartConfigDone();

protected:
    static bool _smartConfigStarted;
    static bool _smartConfigDone;
    static void _smartConfigCallback(uint32_t status, void* result);
```

It does not crash or anything, but when trying to use ESP Smart Config using the app on PlayStore, it gives me Esptouch fail error, after several seconds of trying.

seopyoon, 2017-01-13:
I accidentally closed the issue.  So I reopened it.

me-no-dev, 2017-01-16:
Why not submit a PR? :) we can merge those things and have your name there

seopyoon, 2017-01-17:
@me-no-dev done!

closed 2017-01-17 by seopyoon