espressif/arduino-esp32 #125: OTA programing wi

DiegoMendoza1, 2017-01-13:


elFallino, 2017-01-24:
I guess not (yet?). The Bootloader does not do any WiFi stuff. You would need to write the necessary things on your own either by patching the bootloader or writing user code that receives data OTA and stores them. But it would require you to change the partition layout e.g. to have a second partition where OTA data is stored.

I think that I read somewhere that the current bootloader is able to handle multiple partitions and to start user code from a partition you like. But still the above mentioned things are required.

evanmj, 2017-01-24:
@DiegoMendoza1 Did you try the ArduinoOTA yet on esp32?  http://esp8266.github.io/Arduino/versions/2.0.0/doc/ota_updates/ota_updates.html

DiegoMendoza1, 2017-01-29:
I tried with the esp8266 code but the libraries give me error when compiling the program

sfowlr, 2017-02-10:
Similar to #78

me-no-dev, 2017-02-10:
soon. there are fixes inline upstream. When all is good, I will implement it. Partitions are already there.

me-no-dev, 2017-03-11:
implemented ;)

closed 2017-03-11 by me-no-dev

blackketter, 2017-03-11:
Excited to try this out.  Can you point me to the documentation?

me-no-dev, 2017-03-11:
https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/readme.md#application-example
Skip the Python part.

blackketter, 2017-03-11:
Ok, thanks @me-no-dev!  

Should I open a separate issue suggesting that this repository include its own documentation?

me-no-dev, 2017-03-11:
please do not open an issue. I am quite aware of the many things missing.

DiegoMendoza1, 2017-03-11:
Thanks @me-no-dev it is a good new, i have a problem the cpu gets halted, i have open an issue #260. 

Thanks in advance

arvindr21, 2017-05-28:
Updated 8266 docs URL: https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/readme.rst