espressif/arduino-esp32 #127: Using a user-defined Python version for systems with other Python versions active

MartyMacGyver, 2017-01-14:
If your default python is not 2.7, you run into trouble pretty quickly (while I have 2.7 installed my default is 3.6).

I found that platform.txt can be changed to allow easier setting of the Python executables. While it's not very extensible (given the inelegant way Arduino parses its settings) it allows the user to manually configure things to suit their dev environment. To that end, I moved the following settings to near the top of the file, e.g. (for my installation):

    name=ESP32 Arduino
    version=0.0.1

    tools.esptool.cmd=python27
    tools.esptool.cmd.windows=C:\Python27\python.exe
    tools.esptool.network_cmd=python27
    tools.esptool.network_cmd.windows=C:\Python27\python.exe
    compiler.python.cmd=python27
    compiler.python.cmd.windows=C:\Python27\python.exe

    ....

It would be far better to have two new vars (one for Unix, one for Windows) that the others reference but... that's not something Arduino seems to allow.

Whether this file is reordered as above or not, it'd be useful to let people know about this in the documentation (and that the editing must be done with Arduino closed - some of these specific vars can change on the fly, but some of them will not). It's pretty easy and effective - I was up and running in no time once I hardcoded the Windows vars as above (Unix ones updated for illustration).

Thoughts?

me-no-dev, 2017-01-16:
this have been an issue for quite some time :) PATH and many things are different while Arduino is compiling and that caused me so much headbanging over the years :)
I can move the definitions towards the top :) though very few people look at the configuration files (or so I think)

MartyMacGyver, 2017-01-16:
Thank you! **Whether or not you move the defs (though I think it's easier to group them this way), updating the docs to mention this is the key.** I'm probably not the only person who will be using this in a more complex dev environment (nor is it always desirable to make Python 2.7 the default for a whole system), and had this been called out I'd have known to edit each cmd for Unix, and each cmd.windows for Windows, for all the ones listed above. Given how fragile Arduino seems to be in these ways, a little direction will go a very long way.

(Would that one could simply have a new menu item where one could simply change the appropriate persistent values in a dialog, but that appears to be beyond the scope of the Arduino framework.)

projectgus, 2017-01-16:
FWIW, the latest esptool.py (in latest esp-idf) supports Python 2.7 & 3.x, so some of this problem can be avoided by upgraded. However there's till the problem of needing pyserial installed in that particular Python environment.

MartyMacGyver, 2017-01-17:
I'm curious what the story is on pyserial. Is there some known issue with it in 3.x?

me-no-dev, 2017-01-19:
There is one single definition for esptool on top of platform.txt now and Windows uses binary form of get.py and esptool.py, so Python and pip no longer matter if you are not using IDF.
I think this issue can be closed :)

MartyMacGyver, 2017-01-22:
It's working a lot better now thanks to the executable versions of `get` and `esptool`! No manual config needed. From that perspective, we can close this.

Further notes:

I noticed that the compiler is a minor bump up from the ones on Espressif's site and in the MSYS2 bundle currently.

Also, the following process works and is a bit simpler and saved installing and configuring the full Windows git:

    # Install Arduino as normal

    # Install MSYS2 from Espressif (just download and extract to C:\ so you get C:\msys32)

    # From the MSYS2 shell:
    cd /c/Users/$User/Documents/Arduino/hardware/espressif  # $USER is automatically set in MSYS2
    git clone https://github.com/espressif/arduino-esp32.git esp32  # If initializing
    cd esp32
    git pull  # If updating arduino-esp32
    cd tools
    ./get
    
    # Then proceed with adding the board in Arduino and building normally

One other note: while this seems to work great for ESP32-DevKitC (though you usually have to use the BOOT button on the board), the ESP-WROVER-KIT can be tricky (you never need to use the BOOT button, but it always ends up stuck in download mode the moment you open the serial monitor... not sure why. It takes several tries to get serial comms actually working normally.)

I'm mentioning that for discussion (as @projectgus is on the thread too) - curious if there's any workarounds on either board for those issues.

me-no-dev, 2017-01-22:
@MartyMacGyver when you say that given board is not properly reset, please specify how you are running things :) Are you compiling/uploading through Arduino IDE, or through IDF or Eclipse, so on :) It matters.
I use both boards and have 0 issues resetting or flashing (on a Mac).

MartyMacGyver, 2017-01-23:
Perhaps this is better discussed on the forum if it's not something obvious - I noted it here because users of this would be more likely to notice the difference when the IDE fails to upload and might be more likely to use serial at all.

It's in Windows, as noted above. Just considering the Arduino 1.8.1 IDE / putty, the Core Board v2 and the ESP-WROVER-KIT act differently upon serial disconnect, and from what I can tell the Core board V2 will never take a flash unless you press the BOOT button. However, when you connect to it via the serial monitor or putty, it doesn't reset into download mode (which is good), while the ESP-WROVER-KIT will sometimes do so.

As I've got a lot more data than this on it now, I'll open a topic on the ESP32 forum for further discussion soon. Right now I'm not seeing an obvious workaround that prevents the board from resetting on serial disconnect.

closed 2017-01-23 by MartyMacGyver