espressif/arduino-esp32 #128: Asynchronous Websockets for ESP32

igrr, 2017-01-15:
_From @williamesp2015 on January 14, 2017 9:21_

I would like to migrate from ESP8266 to ESP32 using Asynchronous Websockets. Is there any changes to enable ESP32 for Asynchronous Websockets server and client?
Thank you

_Copied from original issue: espressif/esp-idf#232_

igrr, 2017-01-15:
_From @negativekelvin on January 14, 2017 10:15_

Are you talking about https://github.com/me-no-dev/ESPAsyncWebServer ?

You should ask me-no-dev, post in that repo, or post in the arduino forum at esp32.com.  This isn't the best place for discussions about items outside the scope of this sdk.

tobozo, 2017-03-10:
Do the recent changes on libwebsocket bring some hope ? https://github.com/warmcat/libwebsockets/search?q=esp32&type=Commits&utf8=%E2%9C%93

copercini, 2017-09-09:
AsyncWebServer for ESP32 was implemented by @me-no-dev here: https://github.com/me-no-dev/ESPAsyncWebServer

AsyncTCP is here: https://github.com/me-no-dev/AsyncTCP

AsyncWebsockets still need to be implemented

me-no-dev, 2017-09-09:
The AsyncWebServer has WebSockets inside ;) No client though

copercini, 2017-09-15:
So, solved!

closed 2017-09-15 by copercini

docteurzoidberg, 2018-11-25:
> 
> 
> The AsyncWebServer has WebSockets inside ;) No client though
asyncwebserver is awesome !
would also really appreciate a websocket client x)

docteurzoidberg, 2018-11-25:
> 
> 
> The AsyncWebServer has WebSockets inside ;) No client though

would love a client x)

jhoughjr, 2019-12-18:
also need an async web socket client.

ildarmustafin, 2020-10-10:
https://techtutorialsx.com/2018/09/11/esp32-arduino-web-server-sending-data-to-javascript-client-via-websocket/