espressif/arduino-esp32 #129: WiFi packet loss

maxosprojects, 2017-01-16:
I'm noticing packet loss on ESP32 Thing (from Sparkfun) on WiFi in infrastructure mode.
The device is, supposedly, connected via 802.11n

I disabled SoftAP mode, but that didn't change anything:
```
  // Set STA mode (disable softAP)
  WiFi.mode(WIFI_MODE_STA);
  //Initiate connection
  WiFi.begin(ssid, pwd);
```

The problem was discovered with TCP and then I ran some tests with ICMP pings.
Here are the results at different intervals between requests. Tests were executed on a wired connection to the AP to which ESP32 is connected via WiFi:

**1s**
200 packets transmitted, 196 received, 2% packet loss, time 199325ms
rtt min/avg/max/mdev = 1.117/16.662/1007.573/72.556 ms, pipe 2

**100ms**
200 packets transmitted, 191 received, 4% packet loss, time 20097ms
rtt min/avg/max/mdev = 1.045/17.414/458.020/48.626 ms, pipe 5

**10ms**
200 packets transmitted, 195 received, 2% packet loss, time 2101ms
rtt min/avg/max/mdev = 0.982/4.113/48.350/5.878 ms, pipe 3

Loss rate varies form test to test.

At the same time, a Raspberry Pi 3 connected to the same AP shows no hiccups, nor a single packet loss.

I suspect problem is in one of the binary libraries pulled in here. How can I track down which one?

igrr, 2017-01-16:
We are tracking a similar issue about packet loss with ESP-IDF WiFi driver internally. I will post an update here once we resolve it.

maxosprojects, 2017-01-16:
Thank you!
Could you please link the issue here?

maxosprojects, 2017-02-18:
Any followup on this?

me-no-dev, 2017-02-23:
can you please check the latest version and report back?

me-no-dev, 2017-03-10:
Closing because of no response

closed 2017-03-10 by me-no-dev

Pgpapa-doodl, 2021-03-19:
Hjj

Pgpapa-doodl, 2021-03-19:
Hello All,
I want to detect the data packet loss in esp32. Any idea how to do that? 
Test setup: temperature sensor-master esp32-slave esp32.
I will send temperature sensor data from temperature sensor to Master esp32 and then using wifi to slave esp32. Then I need to check data packet loss when I receive data back from slave to master. 

I am a newbie in this field. Kindly cooperate.

Thank you:)

lbernstone, 2021-03-19:
TCP/IP has packet loss detection and correction built in.  Please don't hijack threads from 3 years ago.  Open a new thread and follow the directions in the [issue template](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md).

Pgpapa-doodl, 2021-03-20:
> TCP/IP has packet loss detection and correction built in. Please don't hijack threads from 3 years ago. Open a new thread and follow the directions in the [issue template](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md).

I want to measure the data packet loss for different intra communication protocol like I2C, SPI and ESP-NOW.
So how can I use TCP/IP for all these protocols?
Also yes I'll keep your advice in mind from next time.

hidogan, 2021-03-20:
Unsubscribe

wachidsusilo, 2022-12-08:
With version 2.0.5, this is still happening. When using WebSocket client, the received WebSocket header has correct payload length, but the data is truncated.

```
// [PacketLength]: [Data]

45: {
  "type": "reg",
  "data": "audio-player"
}
45:                             -> PACKET LOSS
45: {
  "type": "reg",
  "data": "audio-player"
}
45: {
  "type": "reg",
  "data": "audio-player"
}
45: {
  "type": "reg",
  "data": "aud                  -> PACKET LOSS
45: {
  "type": "reg",
  "data": "audio-player"
}
45: {
  "type": "reg",
  "data": "audio-player"
}
45:                             -> PACKET LOSS
45: {
  "type": "reg",
  "data": "audio-player"
}
45: {
  "type": "reg",
  "                             -> PACKET LOSS
45: {
  "type": "reg",
  "data": "audio-player"
}
45: {
  "type": "reg",
  "data": "audio-player"
}
```

And this is the implementation of the WebSocket client for reading incoming packet:
```c++
if (!client || !client->available()) return;

uint16_t bin = 0;
int i = 0;

while (isConnected()) {
    int res = client->read();
    if (res >= 0) bin |= ((uint8_t)res) << i++ * 8;
    if (i > 1) break;
}

Frame::Header header(bin);
uint16_t payloadLen = header.payload;

if ((useMask && header.mask) || (!useMask && !header.mask) || !Frame::isValid(Frame::Opcode(header.opcode))) {
    _close(CloseReason_ProtocolError);
    return;
}

if (header.opcode == 0) {
    _close(CloseReason_UnsupportedData);
    return;
}

if (header.payload == 127) {
    _close(CloseReason_MessageTooBig);
    return;
}

if (header.payload == 126) {
    bin = 0;
    i = 0;
    while (isConnected()) {
        int res = client->read();
        if (res >= 0) bin |= ((uint8_t)res) << i++ * 8;
        if (i > 1) break;
    }
    payloadLen = Crypto::swapEndianness(bin);
}

uint8_t maskingKey[4];
if (header.mask) {
    int res = client->read(maskingKey, 4);
}

char payload[payloadLen + 1];
payload[payloadLen] = 0;
uint16_t read = 0;
uint32_t start = millis();

while (isConnected() && read < payloadLen) {
    int res = client->read((uint8_t*)payload + read, payloadLen - read);
    if (res > 0) read += res;
    if (millis() - start > 1000) {
        break;
    }
}

String payloadData(payload);
if (header.mask) Crypto::remaskData(payloadData, maskingKey);
Serial.println(String(payloadLen) + ": " + payloadData);
```