espressif/arduino-esp32 #132: Does Bluetooth works with this IDE?

MisterBE2, 2017-01-16:
Hello,
does bluetooth works on this platform?

elFallino, 2017-01-17:
As far as I know (I am waiting for them also), BT is not yet implemented within esp32-arduino but since BT is already implemented in esp-idf you can call the esp-idf functions directly "from arduino".

torntrousers, 2017-01-20:
> ...you can call the esp-idf functions directly "from arduino"

Can anyone give an Arduino sketch showing that? Something like the ESP32 picking up advertisements from peripheral nodes?

Idan-BM, 2017-01-26:
> ...you can call the esp-idf functions directly "from arduino"

I'm really looking forward for this capability...
How do you know that the esp-idf commands can be called from the arduino IDE?

elFallino, 2017-01-26:
Because I am doing that, too (not for BT but NV Memory storage, SD Card and things).
The IDF libs are packed with arduino-esp32 because it uses them, too. Like many things in the arduino world, arduino-esp32 is partially "just" an abstraction layer to a more basic set of libraries.
That does not mean that it isn't a great work anyway ;)

Just take a look to the examples within the IDF git. In most cases you can do the same in Arduino as it is done there.

drony, 2017-02-06:
me also wating bluetooth capabilities

WALLTECH, 2017-02-09:
An example of this would be much appreciated

WALLTECH, 2017-02-16:
While this is in development, being able to use the IDF from the Arduino IDE for currently not supported functions is quite important and seemingly easy from plenty of other posts but there isn't an example of how to do it and I've tried with every not found error you could imagine. How would you do this?

me-no-dev, 2017-02-16:
https://gist.github.com/me-no-dev/00bc1a4765350e46992bfd9d5d6c1473

me-no-dev, 2017-02-16:
this is horrible example and will never be in the Arduino section, nor you should expect it to work in the future, but it's what you ask. This is how you integrate IDF BT example into Arduino

WALLTECH, 2017-02-17:
Alright, found this before and remember it not working but I'll try again and see if I can replace parts with the scanning functionality that I need. I'll analyze it. Thank you! Good luck

WALLTECH, 2017-02-19:
I made the changes to the gatt client example in the IDF and set it up in the IDE and it compiles but when it runs there is this error "E (4930) BT: bta_dm_co_ble_load_local_keys: func not ported". Do I have to just wait until BLE is ready?

me-no-dev, 2017-02-19:
Maybe ask this question in IDF?

torntrousers, 2017-02-22:
Thanks for the ESP32_BLE_ADV_Arduino.ino sketch, I gave it a try and with a BLE Scanner app on my phone can see the "ESP-BLE-HELLO" advertisements. Great!

I'd really like to do the other way around though and have the ESP32 picking up advertisements from peripheral nodes. Can anyone help with an example sketch to do that?

me-no-dev, 2017-02-22:
Example and simple class based loosely on the sketch above is not in the main repository. The above sketch no longer works. Closing this issue now ;)

closed 2017-02-22 by me-no-dev

WALLTECH, 2017-02-22:
Thank you very much! Looking forward to scanning examples and central work!

filipeaguiar22, 2017-04-07:
Hi Guys,
  I am really looking forward for the BT implementation !
  I appreciate your effort and look forward to use it on ESP32 board :)