espressif/arduino-esp32 #133: Broken build process

Thoma-HauC, 2017-01-16:
Since last update 3b874d5, I am not able to build my software.
Something goes wrong during the linking.

>Linking everything together...
"/home/user/sketchbook/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc" -nostdlib "-L/home/user/sketchbook/hardware/espressif/esp32/tools/sdk/lib" "-L/home/user/sketchbook/hardware/espressif/esp32/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority  -Wl,--start-group "/tmp/arduino_build_915313/sketch/sketch_jan01a.ino.cpp.o" "/tmp/arduino_build_915313/libraries/WiFi/WiFi.cpp.o" "/tmp/arduino_build_915313/libraries/WiFi/WiFiAP.cpp.o" "/tmp/arduino_build_915313/libraries/WiFi/WiFiClient.cpp.o" "/tmp/arduino_build_915313/libraries/WiFi/WiFiGeneric.cpp.o" "/tmp/arduino_build_915313/libraries/WiFi/WiFiMulti.cpp.o" "/tmp/arduino_build_915313/libraries/WiFi/WiFiSTA.cpp.o" "/tmp/arduino_build_915313/libraries/WiFi/WiFiScan.cpp.o" "/tmp/arduino_build_915313/libraries/WiFi/WiFiServer.cpp.o" "/tmp/arduino_build_915313/libraries/WiFi/WiFiUdp.cpp.o" "/tmp/arduino_build_915313/arduino.ar" -lapp_update -lbootloader_support -lbt -lbtdm_app -lc -lc_nano -lcoap -lcoexist -lcore -lcxx -ldriver -lesp32 -lethernet -lexpat -lfatfs -lfreertos -lhal -ljson -llog -llwip -lm -lmbedtls -lmdns -lmicro-ecc -lnet80211 -lnewlib -lnghttp -lnvs_flash -lopenssl -lphy -lpp -lrtc -lsdmmc -lsmartconfig -lspi_flash -ltcpip_adapter -lulp -lvfs -lwpa -lwpa2 -lwpa_supplicant -lwps -lxtensa-debug-module  -lgcc -lstdc++ -Wl,--end-group -Wl,-EL -o "/tmp/arduino_build_915313/sketch_jan01a.ino.elf"
Multiple libraries were found for "WiFi.h"
 Used: /home/user/sketchbook/hardware/espressif/esp32/libraries/WiFi
 Not used: /opt/arduino-1.6.13/libraries/WiFi
/home/user/sketchbook/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5/../../../../xtensa-esp32-elf/lib/libstdc++.a(vterminate.o):(.literal._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x8): undefined reference to `_impure_ptr'
collect2: error: ld returned 1 exit status
Using library WiFi at version 1.0 in folder: /home/user/sketchbook/hardware/espressif/esp32/libraries/WiFi 
exit status 1
Error compiling for board ESP32 Dev Module.

Thoma

me-no-dev, 2017-01-16:
there is a new toolchain. run tools/get.py again to update it. I hope to fix your problem :)

Thoma-HauC, 2017-01-16:
Fine, it works.

Thoma

closed 2017-01-16 by Thoma-HauC

manitou48, 2017-03-21:
I'm trying to build for ESP32 on  ubuntu with arduino IDE.   The get.py  script failed with dirname ._esptool. I managed to hack the regex to get that to work, i think.  But then the link failed with  undefined reference to `_impure_ptr'  ... UGH.  :-(  

please advise.

me-no-dev, 2017-03-21:
Will fix :) extract tools/dist/esptool-fe69994-linux32.tar.gz into tools (will extract esptool) and it should be fine from then on. Issue is OS X system file that is left in the archive (Mac user here)

manitou48, 2017-03-21:
what about the arduino IDE build error on ubuntu32 building for sparkfun esp32 thing: 
/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/4.8.5/../../../../xtensa-esp32-elf/lib/libstdc++.a(vterminate.o):(.literal._ZN9__gnu_cxx27__verbose_terminate_handlerEv+0x8): undefined reference to `_impure_ptr'

I also tried the arduino IDE build on 64-bit ubuntu 16.04, but there compile won't even start, get:
espressif: Unknown package
??

easyfm, 2017-05-02:
Hello
also error in fresh installed arduino with instruction for ESP32
undefined reference to `_impure_ptr'