espressif/arduino-esp32 #134: Windows fails get.py to download toolchain

hoopsurfer, 2017-01-16:
Needed to comment out "import requests" in get.py to allow it to run on Windows 10.   Seems this import was added recently.  Note current get.py generates the following error:

Traceback (most recent call last):
  File "get.py", line 18, in <module>
    import requests
ImportError: No module named requests

hicotton02, 2017-01-16:
I had the same issue, just install the requests module "pip install requests" from the git bash.

hoopsurfer, 2017-01-16:
Appears "requests" was not needed on Windows, not sure about other platforms.  Either remove the import or update the Windows install instructions with:

$ python -m pip install --upgrade pip && \
> pip install pyserial && \
> pip install requests

me-no-dev, 2017-01-16:
ant to me it seems that Windows is a hit and miss situation. I am not a Windows user but did spend on a license and all and run clean installs on 32 and 64 bit windows 10 machines with 0 issues, but then many people complain about certificate not validated for download or some other mishap while installing...
what is the correct recipe to make it work for all?

hoopsurfer, 2017-01-18:
Is "import requests" doing anything in get.py?   If not then remove that line from get.py and it will fix this, otherwise tweak the Windows instructions to say:  "pip install pyserial requests" and this issue is closed.

tobozo, 2017-01-18:
run `pip install requests`, it's there to handle installations where Cygwin libs are missing

me-no-dev, 2017-01-19:
And it's all to binary for Windows now, so no more Python or pip required :)
Closing this issue!

closed 2017-01-19 by me-no-dev