espressif/arduino-esp32 #135: Arduino IDE 1.8.1 error compiling anything for esp32

hicotton02, 2017-01-16:
followed the directions (today) here:
[https://github.com/espressif/arduino-esp32/blob/master/doc/windows.md](url)

and tried to verify/compile a range of basic sketches. all throw the same error. I am using the ESP32 Thing by sparkfun.

One variance is that I had to manually install the requests module for python before i could run `python get.py`

Error:
Arduino: 1.8.1 (Windows 10), Board: "ESP32 Dev Module, 80MHz, 921600"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\skaz\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\skaz\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:FlashFreq=80,UploadSpeed=921600 -ide-version=10801 -build-path C:\Users\skaz\AppData\Local\Temp\arduino_build_893527 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\skaz\Documents\Arduino\Brewing_Controller\Brewing_Controller.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\skaz\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\skaz\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:FlashFreq=80,UploadSpeed=921600 -ide-version=10801 -build-path C:\Users\skaz\AppData\Local\Temp\arduino_build_893527 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\skaz\Documents\Arduino\Brewing_Controller\Brewing_Controller.ino
Using board 'esp32' from platform in folder: C:\Users\skaz\Documents\Arduino\hardware\espressif\esp32
Using core 'esp32' from platform in folder: C:\Users\skaz\Documents\Arduino\hardware\espressif\esp32
Detecting libraries used...
"C:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/sdmmc" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/coap" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -fno-rtti -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10801 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32  -DESP32 "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32\cores\esp32" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32\variants\esp32" "C:\Users\skaz\AppData\Local\Temp\arduino_build_893527\sketch\Brewing_Controller.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/sdmmc" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/coap" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -fno-rtti -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10801 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32  -DESP32 "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32\cores\esp32" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32\variants\esp32" "C:\Users\skaz\AppData\Local\Temp\arduino_build_893527\sketch\Brewing_Controller.ino.cpp" -o "C:\Users\skaz\AppData\Local\Temp\arduino_build_893527\preproc\ctags_target_for_gcc_minus_e.cpp"
**exit status -1073741515**
**Error compiling for board ESP32 Dev Module.**

hoopsurfer, 2017-01-16:
+1 
Arduino: 1.8.1 (Windows Store 1.8.1.0) (Windows 10), Board: "ESP32 Dev Module, 80MHz, 921600"
exit status -1073741515
Error compiling for board ESP32 Dev Module.

me-no-dev, 2017-01-16:
please rerun tools/get.py to get an updated toolchain. And please report if that fixed it :)

hicotton02, 2017-01-16:
reran the command and same error.

hicotton02, 2017-01-16:
took it a step further and deleted everything reran whole setup:

`$ mkdir -p ~/Documents/Arduino/hardware/espressif && \
> cd ~/Documents/Arduino/hardware/espressif && \
> git clone https://github.com/espressif/arduino-esp32.git esp32 && \
> cd esp32/tools/ && \
> python get.py
Cloning into 'esp32'...
remote: Counting objects: 2529, done.
remote: Compressing objects: 100% (195/195), done.
remote: Total 2529 (delta 22), reused 0 (delta 0), pack-reused 2315
Receiving objects: 100% (2529/2529), 49.21 MiB | 15.67 MiB/s, done.
Resolving deltas: 100% (937/937), done.
Checking out files: 100% (884/884), done.
Platform: i686-mingw32
Downloading xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0.zip
Done
Extracting dist/xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0.zip`

Error:

`Arduino: 1.8.1 (Windows 10), Board: "ESP32 Dev Module, 80MHz, 921600"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\skaz\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\skaz\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:FlashFreq=80,UploadSpeed=921600 -ide-version=10801 -build-path C:\Users\skaz\AppData\Local\Temp\arduino_build_534755 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\skaz\Documents\Arduino\Brewing_Controller\Brewing_Controller.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\skaz\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\skaz\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:FlashFreq=80,UploadSpeed=921600 -ide-version=10801 -build-path C:\Users\skaz\AppData\Local\Temp\arduino_build_534755 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\skaz\Documents\Arduino\Brewing_Controller\Brewing_Controller.ino
Using board 'esp32' from platform in folder: C:\Users\skaz\Documents\Arduino\hardware\espressif\esp32
Using core 'esp32' from platform in folder: C:\Users\skaz\Documents\Arduino\hardware\espressif\esp32
Detecting libraries used...
"C:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/sdmmc" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/coap" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -fno-rtti -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10801 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32  -DESP32 "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32\cores\esp32" "-IC:\Users\skaz\Documents\Ard [... cut ...]  -E -CC -DF_CPU=240000000L -DARDUINO=10801 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32  -DESP32 "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32\cores\esp32" "-IC:\Users\skaz\Documents\Arduino\hardware\espressif\esp32\variants\esp32" "C:\Users\skaz\AppData\Local\Temp\arduino_build_534755\sketch\Brewing_Controller.ino.cpp" -o "C:\Users\skaz\AppData\Local\Temp\arduino_build_534755\preproc\ctags_target_for_gcc_minus_e.cpp"
exit status -1073741515
Error compiling for board ESP32 Dev Module.`

hoopsurfer, 2017-01-16:
+1 Cleaned, Rebooted, Reran:
Cloning into 'esp32'...
remote: Counting objects: 2529, done.
remote: Compressing objects: 100% (195/195), done.
remote: Total 2529 (delta 22), reused 0 (delta 0), pack-reused 2315
Receiving objects: 100% (2529/2529), 49.21 MiB | 4.25 MiB/s, done.
Resolving deltas: 100% (937/937), done.
Checking out files: 100% (884/884), done.
Platform: i686-mingw32
Downloading xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0.zip
Done
Extracting dist/xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0.zip
...
exit status -1073741515
Error compiling for board ESP32 Dev Module.

me-no-dev, 2017-01-16:
seems that I need to test this on Windows.. maybe the new toolchain or something. no issues on a Mac.

hicotton02, 2017-01-16:
let me know if there is something i can help with. note all my installs are 64 bit. (Arduino, Python, etc)

porkyneal, 2017-01-16:
+1, just received the module and have the same error message. (Win 10, X64. Arduino 1.6.10 hourly Build)

tolson2000, 2017-01-16:
Confirmed to work OK with Arduino 1.8.1 on Linux SuSE 13.2.

me-no-dev, 2017-01-16:
you windows guy can try the old toolchain from [here](https://dl.espressif.com/dl/xtensa-esp32-elf-win32-1.22.0-59.zip) and please do report if working

me-no-dev, 2017-01-16:
if it complains about missing impure_ptr, add this to your sketch:
```cpp

void *operator new(size_t size)
{
    return malloc(size);
}

void *operator new[](size_t size)
{
    return malloc(size);
}

void operator delete(void * ptr)
{
    free(ptr);
}

void operator delete[](void * ptr)
{
    free(ptr);
}

extern "C" void __cxa_pure_virtual(void) __attribute__ ((__noreturn__));
extern "C" void __cxa_deleted_virtual(void) __attribute__ ((__noreturn__));

void __cxa_pure_virtual(void)
{
    abort();
}

void __cxa_deleted_virtual(void)
{
    abort();
}
```

SpawnY0815, 2017-01-16:
Last two posts solves my problem on Windows 10 64bit, Arduino IDE 1.8.1 64bit and python 2.7.12 64bit

thx me-no-dev

robertwatsonbath, 2017-01-16:
Older toolchain and and impure_ptr fix works for me too (Win 7 64bit, Arduino 1.8.0, python 2.7.12)
Thanks me-no-dev.

hicotton02, 2017-01-16:
confirmed. with the impure_ptr fix.

hoopsurfer, 2017-01-17:
+1 works with older toolchain and required impure_ptr fix.  Also I had to hold button to download after 10 attempts.  Seems esp_tool.py may be more touchy connecting to dev module than in the past.  
Arduino: 1.8.1 (Windows Store 1.8.1.0) (Windows 10), Board: "ESP32 Dev Module, 80MHz, 921600", Python 2.7.13 (with added pip install requests)

me-no-dev, 2017-01-19:
alright guys :) toolchain updated and new get.exe added next to get.py. doubleclick get.exe to update the toolchain

hoopsurfer, 2017-01-19:
Clean clone then get.exe instead of get.py and all works fine 1) I see updated Windows directions so no concerns over import requests, 2) no issue with compiler error.  3) no more issue with impure_ptr.   Looks good - thanks.

hicotton02, 2017-01-19:
Confirmed good sir. looks good!

closed 2017-01-19 by hicotton02

me-no-dev, 2017-01-19:
Awesome :) now let's add binary esptool so users would have only git as dependency

me-no-dev, 2017-01-19:
@hoopsurfer I also noticed on Windows esptool to have harder time to reset the board. Will ask around

MrJustPaul, 2017-07-17:
> alright guys :) toolchain updated and new get.exe added next to get.py. Git pull the repo and doubleclick get.exe to update the toolchain
> 
Thank you me-no-dev!  It looks like you have the fix!  But I'm lost.  Sorry - total noob to Arduino/ESP32/Git/Python/...   I downloaded and extracted "xtensa-esp32-elf-win32-1.22.0-59 (1).zip". Now What? Can you explain the above for a slow learner?

jjimmy-S, 2018-03-27:
@me-no-dev  I'm getting the same error with both the new toolchain and the old one that you posted, I'm running Win 8.1 x64, Arduino 1.8.5, and the esp 32 dev kit v1 by "do it" this is the error:

exit status 1
Error compiling for board ESP32 Dev Module.

Do you have any suggestions?

beegee-tokyo, 2018-03-27:
@jjimmy-S after recommended installation, in the last where you have to run get.exe, did it throw any error message? I think in your case the toolchain was not installed. If get.exe throws errors you can [this](https://desire.giesecke.tk/index.php/2018/03/10/esp32-framework-installation-on-windows-xp/).

Or you can install the toolchain and other tools manually.
Download 

1. [toolchain](https://dl.espressif.com/dl/xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip)
2. [esptool](https://dl.espressif.com/dl/esptool-4dab24e-windows.zip)
3. [mkspiffs](https://github.com/igrr/mkspiffs/releases/download/0.2.1/mkspiffs-0.2.1-windows.zip)    

and unzip 

1. content of xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip **into** SKETCHBOOK_PATH\hardware\espressif\esp32\tools\xtensa-esp32-elf
2. esptool.exe from esptool-4dab24e-windows.zip **to** SKETCHBOOK_PATH\hardware\espressif\esp32\tools\
3. mkspiffs.exe from mkspiffs-0.2.1-windows.zip **into** SKETCHBOOK_PATH\hardware\espressif\esp32\tools\mkspiffs

Cysign, 2018-05-07:
Same here. Tried to compile FastLed examples.

`In file included from C:\ProgrammeKonsistent\arduino-1.8.1\libraries\FastLED/led_sysdefs.h:27:0,

                 from C:\ProgrammeKonsistent\arduino-1.8.1\libraries\FastLED/FastLED.h:44,

                 from C:\Users\cysig\AppData\Local\Temp\arduino_modified_sketch_37868\DemoReel100.ino:1:

C:\ProgrammeKonsistent\arduino-1.8.1\libraries\FastLED/platforms/avr/led_sysdefs_avr.h:12:20: fatal error: avr/io.h: No such file or directory

compilation terminated.

Mehrere Bibliotheken wurden für "FastLED.h" gefunden
 Benutzt: C:\ProgrammeKonsistent\arduino-1.8.1\libraries\FastLED
 Nicht benutzt: C:\ProgrammeKonsistent\arduino-1.8.1\libraries\FastLED-3.1.0
exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.`

1TekPro, 2019-01-26:
\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++": file does not exist
this is what I always get after trying all the fixes.

Eddiiie, 2020-04-16:
why is the path screwed up?
\  and /

lbernstone, 2020-04-16:
Windows uses \, unix uses /
If you are on a windows machine, seeing backslashes is a good indicator the problem is on your system.  forward slashes would mean the error is in something compiled on a development system.