espressif/arduino-esp32 #137: Build Error on El Capitan 10.11.6

forthlightning, 2017-01-18:
Hi, 

Using a fresh pull of arduino-esp32, and just ran /tools/get.py to get a new toolchain.

Here is the result of "make flash":

`
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o mconf.o mconf.c
flex -L -P zconf -o zconf.lex.c zconf.l
zconf.l:255: warning, -s option given but default rule can be matched
bison -t -l -p zconf -o zconf.tab.c zconf.y
sed -E "s/\\x0D$//" zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o zconf.tab.o zconf.tab.c
lxdialog/check-lxdialog.sh -check cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -lncurses 
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/checklist.o lxdialog/checklist.c
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/util.o lxdialog/util.c
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/inputbox.o lxdialog/inputbox.c
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/textbox.o lxdialog/textbox.c
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/yesno.o lxdialog/yesno.c
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/menubox.o lxdialog/menubox.c
cc -o mconf mconf.o zconf.tab.o lxdialog/checklist.o lxdialog/util.o lxdialog/inputbox.o lxdialog/textbox.o lxdialog/yesno.o lxdialog/menubox.o -lncurses 
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o conf.o conf.c
cc -o conf conf.o  zconf.tab.o -lncurses 
GENCONFIG
CC src/bootloader_flash.o
CC src/bootloader_random.o
CC src/efuse.o
CC src/esp_image_format.o
CC src/flash_encrypt.o
CC src/flash_partitions.o
CC src/secure_boot.o
CC src/secure_boot_signatures.o
AR libbootloader_support.a
CC log.o
AR liblog.a
CC spi_flash_rom_patch.o
AR libspi_flash.a
CC micro-ecc/uECC.o
AR libmicro-ecc.a
CC bootloader_start.o
AR libmain.a
LD bootloader.elf
esptool.py v2.0-dev
Firmware/esp-idf/components/partition_table/partitions_singleapp.csv...
CC esp_ota_ops.o
AR libapp_update.a
CC cores/esp32/libb64/cdecode.o
/components/arduino/cores/esp32/libb64/cdecode.c: In function 'base64_decode_value':
/components/arduino/cores/esp32/libb64/cdecode.c:15:5: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     if (value_in < 0 || value_in > decoding_size) {
     ^
/components/arduino/cores/esp32/libb64/cdecode.c: In function 'base64_decode_block':
/components/arduino/cores/esp32/libb64/cdecode.c:45:13: warning: comparison is always false due to limited range of data type [-Wtype-limits]
             } while (fragment < 0);
             ^
/components/arduino/cores/esp32/libb64/cdecode.c:55:13: warning: comparison is always false due to limited range of data type [-Wtype-limits]
             } while (fragment < 0);
             ^
/components/arduino/cores/esp32/libb64/cdecode.c:66:13: warning: comparison is always false due to limited range of data type [-Wtype-limits]
             } while (fragment < 0);
             ^
components/arduino/cores/esp32/libb64/cdecode.c:77:13: warning: comparison is always false due to limited range of data type [-Wtype-limits]
             } while (fragment < 0);
             ^
CC cores/esp32/libb64/cencode.o
CC cores/esp32/esp32-hal-adc.o
CC cores/esp32/esp32-hal-dac.o
CC cores/esp32/esp32-hal-gpio.o
CC cores/esp32/esp32-hal-i2c.o
CC cores/esp32/esp32-hal-ledc.o
CC cores/esp32/esp32-hal-matrix.o
CC cores/esp32/esp32-hal-misc.o
CC cores/esp32/esp32-hal-sigmadelta.o
CC cores/esp32/esp32-hal-spi.o
components/arduino/cores/esp32/esp32-hal-spi.c: In function 'spiAttachSCK':
components/arduino/cores/esp32/esp32-hal-spi.c:30:103: error: 'VSPICLK_OUT_IDX' undeclared (first use in this function)
 #define SPI_CLK_IDX(p)  ((p==0)?SPICLK_OUT_IDX:((p==1)?SPICLK_OUT_IDX:((p==2)?HSPICLK_OUT_IDX:((p==3)?VSPICLK_OUT_IDX:0))))
                                                                                                       ^
components/arduino/cores/esp32/esp32-hal-spi.c:87:29: note: in expansion of macro 'SPI_CLK_IDX'
     pinMatrixOutAttach(sck, SPI_CLK_IDX(spi->num), false, false);
                             ^
components/arduino/cores/esp32/esp32-hal-spi.c:30:103: note: each undeclared identifier is reported only once for each function it appears in
 #define SPI_CLK_IDX(p)  ((p==0)?SPICLK_OUT_IDX:((p==1)?SPICLK_OUT_IDX:((p==2)?HSPICLK_OUT_IDX:((p==3)?VSPICLK_OUT_IDX:0))))
                                                                                                       ^
components/arduino/cores/esp32/esp32-hal-spi.c:87:29: note: in expansion of macro 'SPI_CLK_IDX'
     pinMatrixOutAttach(sck, SPI_CLK_IDX(spi->num), false, false);
                             ^
make[1]: *** [cores/esp32/esp32-hal-spi.o] Error 1
make: *** [arduino-build] Error 2
`

forthlightning, 2017-01-18:
If i revert to 6edf8a7 the build completes.

me-no-dev, 2017-01-19:
this was fixed and merged yesterday. Please pull and confirm. Also to note that if you are running "make flash" that means you run through IDF and "python get.py" will do you no good (it's for Arduino IDE only)

forthlightning, 2017-01-19:
Ok, this clarification re IDF and Arduino is helpful. Thanks!

If I am running arduino as component, do both the esp32 core in the arduino component and the IDF stored in IDF PATH get compiled?

I pulled a fresh copy of both the IDF and the arduino component, and I still get the same messages from make flash.

forthlightning, 2017-01-19:
JK this was a build error on my end. Had to update toolchain.

closed 2017-01-19 by forthlightning