espressif/arduino-esp32 #138: Udp endPacket() returns -1 on error and 7 on success

tonu42, 2017-01-18:
Been using Udp a lot and believe when the packet fails to send successfully it should return 0, and on success a 1.

Below screen shots to support. 

[image]

[image]

me-no-dev, 2017-01-19:
thanks for noting :) fixed in the latest commit.

closed 2017-01-19 by me-no-dev

tonu42, 2017-01-19:
Glad to have helped. Glad I was not mistaken. This was my first ever issue I reported on Github! Trying to be involved with the ESP32 development to make it better for everyone. Hope you did not think I was nitpicking.

me-no-dev, 2017-01-19:
no worries :) things like that are important and easy to miss at the same time