espressif/arduino-esp32 #141: SSL Issue with get.exe under windows behind corporate firewall

mindforger, 2017-01-23:
>Platform: i686-mingw32
Downloading xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0-2.zip
Traceback (most recent call last):
  File "get.py", line 138, in <module>
    get_tool(tool)
  File "get.py", line 88, in get_tool
    r = requests.get(url,verify=False)
  File "site-packages\requests\api.py", line 70, in get
  File "site-packages\requests\api.py", line 56, in request
  File "site-packages\requests\sessions.py", line 488, in request
  File "site-packages\requests\sessions.py", line 609, in send
  File "site-packages\requests\adapters.py", line 497, in send
requests.exceptions.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)
Failed to execute script get

main reason is an corporate root certificate used to reencrypt SSL connections because of virus live scanning

i got an .cer file from our IT and no clue on how to pass it to the get.exe

me-no-dev, 2017-01-23:
your best bet is to probably directly download the files and extract then into tools/.
- [Toolchain](https://dl.espressif.com/dl/xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0-2.zip)
- [EspTool](https://dl.espressif.com/dl/esptool-fe69994-windows.zip)

mindforger, 2017-01-23:
thank you :)

i've managed to make a java certificate patcher but i have no idea how i could get the certificate into the crosscompiled get.exe

closed 2017-01-23 by mindforger