espressif/arduino-esp32 #142: Error sending UDP packets

acoulson2000, 2017-01-23:
I'm getting an error with a little test program to send UDP packets, I want to use Multicast, but I get the same error just using a broadcast IP.

Here is my sketch:
`/*
 * Copyright (c) 2015, Majenko Technologies
 * All rights reserved.
 * 
 * Redistribution and use in source and binary forms, with or without modification,
 * are permitted provided that the following conditions are met:
 * 
 * * Redistributions of source code must retain the above copyright notice, this
 *   list of conditions and the following disclaimer.
 * 
 * * Redistributions in binary form must reproduce the above copyright notice, this
 *   list of conditions and the following disclaimer in the documentation and/or
 *   other materials provided with the distribution.
 * 
 * * Neither the name of Majenko Technologies nor the names of its
 *   contributors may be used to endorse or promote products derived from
 *   this software without specific prior written permission.
 * 
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND
 * ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED
 * WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
 * DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE FOR
 * ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES
 * (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
 * LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON
 * ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
 * (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
 * SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 */

/* Create a WiFi access point and provide a web server on it. */

#include <WiFi.h>
#include <WiFiUDP.h>

const char *ssid = "coulson2";
const char *password = "f00f00f00";
unsigned int portMulti = 5000;      // local port to send UDP packets on

WiFiUDP Udp;

IPAddress     broadcastIP(192, 168, 255, 255);
//IPAddress ipMulti(239, 0, 0, 57);
IPAddress myIP;

boolean lightsOn = false;        
unsigned long lastTime = millis();          

byte red[] = {0xFF, 0x00, 0x00, 0x00};
byte black[] = {0x00, 0x00, 0x00, 0x00};
byte *buff;

void setup() {
	delay(1000);
	Serial.begin(115200);
	Serial.println();

  // Station mode
  Serial.println("** Scan Networks **");
  byte numSsid = WiFi.scanNetworks();
  Serial.println("SSID List:");
  Serial.println(numSsid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  int tries=0;
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    tries++;
    if (tries > 30){
      break;
    }
  }
  Serial.println("Connected!");
  delay(3000);
  myIP = WiFi.localIP();
  
  
  Serial.print("My IP address: ");
  Serial.println(myIP);
  Serial.println("done with setup");
}

void loop() {
  //Serial.print("My IP address: ");
  //Serial.println(myIP);
  //  Udp.beginPacketMulticast(ipMulti, portMulti, WiFi.localIP());
  Udp.beginPacket(broadcastIP, portMulti);
  if (lightsOn) {
    Serial.println("off");
    byte tmp[] = {255, 0, 0, 0};
    Udp.write(tmp, 4); 
  } else {
    Serial.println("on");
    byte tmp[] = {255, 254, 0, 0};
    Udp.write(tmp, 4); 
  }
  Udp.endPacket();
  if( (millis() - lastTime) > 500 ) {
    lastTime = millis();
    if (lightsOn) {
      lightsOn = false;
    } else {
      lightsOn = true;
    }
  }
  delay(100);
  yield();
}
`

and here is my output:
`** Scan Networks **
SSID List:
10
.......Connected!
My IP address: 192.168.2.72
done with setup
on
Guru Meditation Error of type StoreProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x4010c03e  PS      : 0x00060a30  A0      : 0x8010c05c  A1      : 0x3ffdc890  
A2      : 0x3ffc251c  A3      : 0x000000ff  A4      : 0x3ffc8ce0  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000001  A11     : 0x3f4011da  A12     : 0x00000002  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000012  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x4010c03e:0x3ffdc890 0x4010c05c:0x3ffdc8b0 0x400d0f89:0x3ffdc8d0 0x40104f35:0x3ffdc900

CPU halted.
`

Thanks for your work on this. Can't wait till it's mature!

me-no-dev, 2017-01-23:
first try to broadcast to 192.168.2.255, then look at [ESPExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder) as a mean to decode that backtrace that you are getting

acoulson2000, 2017-01-24:
Ok, so apparently with esp32, unlike the 8266 sdk, you must first call one of the Udp.begin() overloads, or Udp.beginMulticast(IPAddress a, uint16_t p) prior to sending packets in order for the transmit buffer to be allocated. 

My modified code, which works with multicast, looks like:

```void setup() {
  // Station mode
  Serial.println("** Scan Networks **");
  byte numSsid = WiFi.scanNetworks();
  Serial.println("SSID List:");
  Serial.println(numSsid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  int tries=0;
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    tries++;
    if (tries > 30){
      break;
    }
  }
  Serial.println("Connected!");
  delay(3000);
  myIP = WiFi.localIP();

  Udp.beginMulticast(ipMulti, portMulti);
}

void loop() {
  Udp.beginMulticastPacket();
  if (lightsOn) {
    Serial.println("off");
    byte tmp[] = {255, 0, 0, 0};
    Udp.write(tmp, 4); 
  } else {
    Serial.println("on");
    byte tmp[] = {255, 254, 0, 0};
    Udp.write(tmp, 4); 
  }
  Udp.endPacket();
  if( (millis() - lastTime) > 500 ) {
    lastTime = millis();
    if (lightsOn) {
      lightsOn = false;
    } else {
      lightsOn = true;
    }
  }
  delay(100);
  yield();
}```

closed 2017-01-24 by acoulson2000