espressif/arduino-esp32 #143: WiFi.config does not work

hcs-svn, 2017-01-23:
```
#include <Arduino.h>
#include <WiFi.h>
#include <esp_system.h>

void setup(void) {
  Serial.begin(57600);
  delay(10);
  Serial.println();

  Serial.println(esp_get_idf_version());

  WiFi.mode(wifi_mode_t::WIFI_MODE_STA);
  WiFi.config(IPAddress(192, 168, 31, 212), IPAddress(192, 168, 31, 1), IPAddress(255, 255, 255, 0));
  WiFi.begin("top", "secret");

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
   Serial.println();
   
   Serial.println("IP address: " + WiFi.localIP().toString());
}
```
Result of this code is:
> v1.0-477-g0865819
> ...........
> IP address: 192.168.31.108

Looks a if it would use allways DHCP, should be 192.168.31.212 and not 192.168.31.108

Is it a bug or do I something wrong?

me-no-dev, 2017-01-23:
It's a bug! thanks for reporting this :)

closed 2017-01-23 by me-no-dev

BrycePeterson, 2017-07-27:
The following issue is present for me when using WiFi.config. 
`
IPAddress chipIP = (192,168,10,22);
WiFi.config(chipIP,WiFi.gatewayIP(),WiFi.subnetMask());
`
Result of code is:
> IP Address: 22.0.0.0

Seems like every number I put in the last parameter of IPAddress is assigned as the first number of the new IP address, all other parameters are discarded.