espressif/arduino-esp32 #144: WiFi.RSSI() returns always zero

hcs-svn, 2017-01-23:
In both modes (STA and AP) the result is always zero.

me-no-dev, 2017-01-23:
hey thanks for noting this! API was not available before and I have not noticed when it was added :)

closed 2017-01-23 by me-no-dev

melvinpmathew, 2017-09-06:
I still see that with AP mode WiFi.RSSI() is 0. It does return from (return info.rssi;). But its always 0. Any idea whats wrong? I am having an ESP32 Wrroom with SoftAp enabled.

dberenguer, 2018-02-28:
Same issue here. Running WiFi in promiscuous mode.

vksgaikwad3, 2021-04-26:
I am facing the same issue with my ESP32 based device, in between the WiFi.RSSI() returns 0 and then continuously return 0, hence it gets disconnected from router. Any idea why it returns 0.

thomastech, 2022-01-06:
I know this issue is closed. But I experienced the same issue:. AP mode RSSI is always 0, STA mode RSSI is OK. 

But after thinking about it I believe this is correct. In STA mode the RSSI is the signal strength from the WiFi router. But in AP mode the ESP device is the signal source. So there's no "Received Strength" from a router. Assigning zero is probably the best value to return.

- Thomas