espressif/arduino-esp32 #147: Watchdog functions missing

cbxbiker61, 2017-01-25:
Could you include task_wdt.c in the build?  It's currently not in the libraries.

me-no-dev, 2017-01-25:
Where is that file? I include everything that comes in esp-idf

cbxbiker61, 2017-01-25:
esp-idf/components/esp32/task_wdt.c

me-no-dev, 2017-01-25:
if it's in idf, it's in the libs :) all libs are compiled and included with arduino

igrr, 2017-01-25:
Task watchdog functions will only be compiled if task watchdog option is enabled in sdkconfig (it's not enabled for Arduino build). Uncheck "also watch idle task on app CPU" option though, otherwise task watchdog will complain about the always-running arduino task.

Curclamas, 2017-02-04:
Is there any possibility to use "make menuconfig"/change sdkconfig while using Arduino IDE? I tried to change the sdconfig.h in the Arduino folder but that did not change anything. Also (even when #include "esp_task_wdt.h") it won't compile since it complains that  "esp_task_wdt_init" is not defined. I also added the wdt config related lines from another project to sdconfig.h manually but it did not change anything.
TL;DR: How could one make use of the task wdt function while using Arduino IDE?

me-no-dev, 2017-02-04:
I'm prepping some more changes and will include it in the next update of the libs.

me-no-dev, 2017-02-05:
can you please pull the latest commits, switch to my branch and see if watchdog is now there :)
```git pull && git checkout me-no-dev```

me-no-dev, 2017-02-08:
watchdog functions added :)

closed 2017-02-08 by me-no-dev

rtrkamalesh, 2018-03-31:
I am new to this ... can you say step by step how to make work the watchdog functions.. ?

rtrkamalesh, 2018-03-31:
I am new to this... can you please instruct me step by step to make watchdog functions work for esp32 ? @me-no-dev