espressif/arduino-esp32 #148: Does usart1 work ?

forhuan, 2017-01-26:
# Code example

```C
HardwareSerial Serial1(1);

void setup() {
    Serial1.begin(115200);
}
```
# Error
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:1760
load:0x40078000,len:6668
load:0x40080000,len:252
entry 0x40080034
Guru Meditation Error of type StoreProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400d0944  PS      : 0x00060530  A0      : 0x800d0c2c  A1      : 0x3ffc9aa0  
A2      : 0x3ffc1090  A3      : 0x00000009  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffc62b0  A8      : 0x00000001  A9      : 0x3ffc9a80  
A10     : 0x3ff440a4  A11     : 0x00000011  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000017  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x000003fd  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d0944:0x3ffc9aa0 0x400d0c2c:0x3ffc9ac0 0x400d06f4:0x3ffc9af0 0x400d0599:0x3ffc9b20 0x400dfa52:0x3ffc9b40

CPU halted.
```

me-no-dev, 2017-01-27:
UART1 defaults to pins 9 and 10 which are used by the onboard flash. Either use UART2 on pins 16 and 17 or define the pins you want to use when starting UART1

forhuan, 2017-01-27:
thank you and i'll try it later

closed 2017-02-08 by me-no-dev