espressif/arduino-esp32 #150: undefined reference to `esp_vfs_fat_sdmmc_unmount()'

luca-saggese, 2017-01-27:
Maybe i'm missing something stupid...
I'm trying to communicate to the sd card on a custom dev board, however when i try to compile the schetch i get: undefined reference to `esp_vfs_fat_sdmmc_unmount()'

here is a sample code:
`#include <stdio.h>
#include <string.h>
#include <sys/unistd.h>
#include <sys/stat.h>
#include "esp_err.h"
#include "esp_log.h"
#include "esp_vfs_fat.h"
#include "driver/sdmmc_host.h"
#include "driver/sdmmc_defs.h"
#include "sdmmc_cmd.h"
//void app_main(void);

void setup()
{
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("test ");
    esp_vfs_fat_sdmmc_unmount();
}

int value = 0;

void loop()
{

}`

Any help?

me-no-dev, 2017-01-27:
put the includes in ```extern "C" {}```

luca-saggese, 2017-01-27:
work perfectly thanks!

closed 2017-01-27 by me-no-dev