espressif/arduino-esp32 #151: Referencing the RST pin

hicotton02, 2017-01-27:
please excuse my ignorance, but is there a way to define the actual reset (EN) pin in code?

for example:
#define TFT_RST EN

I am trying to get my ili9341 to work (code wont compile) and it is hardwired to the reset pin.

me-no-dev, 2017-01-27:
I'm not sure what reset pin you are talking about? ESP32's reset pin? Or reset on the screen? How is that screen connected? Board? Library?

hicotton02, 2017-01-27:
Im sorry. it was clearer in my head. I have an ESP32 Thing from Sparkfun and I wired it's reset pin to the reset pin of the Adafruit ILI9341 touchscreen. I am trying to instanciate the class:

hicotton02, 2017-01-27:
seems my issues is a little deeper: just assigned it to Null to check, and got all of this:

`C:\Users\skaz\Documents\Arduino\libraries\Adafruit_ILI9341\Adafruit_ILI9341.cpp: In member function 'void Adafruit_ILI9341::spiwrite(uint8_t)':

C:\Users\skaz\Documents\Arduino\libraries\Adafruit_ILI9341\Adafruit_ILI9341.cpp:113:3: error: 'mosiport' was not declared in this scope

  *mosiport |=  mosipinmask;

   ^

C:\Users\skaz\Documents\Arduino\libraries\Adafruit_ILI9341\Adafruit_ILI9341.cpp:113:16: error: 'mosipinmask' was not declared in this scope

  *mosiport |=  mosipinmask;

                ^

C:\Users\skaz\Documents\Arduino\libraries\Adafruit_ILI9341\Adafruit_ILI9341.cpp:116:3: error: 'mosiport' was not declared in this scope

  *mosiport &= ~mosipinmask;

   ^

C:\Users\skaz\Documents\Arduino\libraries\Adafruit_ILI9341\Adafruit_ILI9341.cpp:116:16: error: 'mosipinmask' was not declared in this scope

  *mosiport &= ~mosipinmask;

                ^

C:\Users\skaz\Documents\Arduino\libraries\Adafruit_ILI9341\Adafruit_ILI9341.cpp:119:8: error: 'clkport' was not declared in this scope

       *clkport |=  clkpinmask;

        ^

C:\Users\skaz\Documents\Arduino\libraries\Adafruit_ILI9341\Adafruit_ILI9341.cpp:119:20: error: 'clkpinmask' was not declared in this scope

       *clkport |=  clkpinmask;

                    ^`

should I just close this issue?

closed 2017-01-27 by hicotton02