espressif/arduino-esp32 #160: Is it possible to connect to WPA2 Enterprise with Arduino IDE

sebasqure, 2017-01-31:
I'm using the Sparkfun ESP32 Thing board.

arcao, 2017-02-09:
ESP32 supports WPA2 enterprise and also IDF supports it, but it's not implemented in Arduino-esp32 yet.

sebasqure, 2017-02-09:
So how can I connect the Sparkfun ESP32 Thing board to a WPA2 enterprise and use it with the Blynk app? Is there any way of doing this or I better wait untill it's implemented?

SpiraMirabilis, 2017-03-09:
Until it is supported on the Arduino IDE you'll have to use Expressif's SDK or better to compile your code that requires WPA 2 enterprise. 
https://github.com/espressif/esp-idf/releases

juliantoledo, 2017-06-30:
Is it still not supported?

martinius96, 2018-06-04:
Working with my sketch!
I have tested it on Eduroam + local 802.1x network!
https://github.com/martinius96/ESP32-Eduroam

naikrovek, 2018-07-30:
Don't close it, yet.  That sketch only works when the WPA2-Enterprise network happens to be setup the way that the ESP32 wants it set up with its default configuration.

The majority of people won't have have that configuration, and the current Arduino software offers no way to change the EAP negotiation method.

gojimmypi, 2020-06-08:
I have enterprise WiFi working with Cisco hardware as shown [here](https://github.com/gojimmypi/RFID_Logger/blob/c1bec2fed8ce11a6ac51ae7970a1d4a9a9855976/WiFiHelper.cpp#L81):

```
WiFi.mode(WIFI_STA); // be sure to set mode FIRST
esp_wifi_sta_wpa2_ent_set_identity((uint8_t*)SECRET_EAP_ID, strlen(SECRET_EAP_ID)); //provide identity
esp_wifi_sta_wpa2_ent_set_username((uint8_t*)SECRET_EAP_USERNAME, strlen(SECRET_EAP_USERNAME)); //provide username
esp_wifi_sta_wpa2_ent_set_password((uint8_t*)SECRET_EAP_PASSWORD, strlen(SECRET_EAP_PASSWORD)); //provide password
esp_wpa2_config_t config = WPA2_CONFIG_INIT_DEFAULT();
esp_wifi_sta_wpa2_ent_enable(&config);
WiFi.begin(SECRET_WIFI_SSID);
```
Note that my `SECRET_EAP_USERNAME` and `SECRET_EAP_ID` are the same value and in the format `DOMAIN\\username`  (Microsoft Active Directory Authentication). I was not able to get `username@fqdn.com` format working as noted in https://github.com/martinius96/ESP32-eduroam/issues/4

I'm using the Visual Micro extension for Visual Studio, but it is essentially the Arduino IDE with a different front end.

igrr, 2022-04-05:
This has been implemented in https://github.com/espressif/arduino-esp32/pull/6398, thanks @jpswensen for the pull request!

closed 2022-04-05 by igrr