espressif/arduino-esp32 #162: wificlient doesn´t work

PeWos, 2017-01-31:
Hi,

I am trying to start the wificlient example, when I open the serial port it shows always "wifi:  reconnect".
I am working with the esp32 development kit from espressif.

me-no-dev, 2017-01-31:
there isn't really much that you are telling here :) please show code, serial output, something that we can look at and troubleshoot. Lots of people are using WiFiClient so it's working :)

PeWos, 2017-01-31:
I'm doing something wrong :( I do not have much knowledge about WiFi.

This is the example code I use:

```
/*
 *  This sketch sends data via HTTP GET requests to data.sparkfun.com service.
 *
 *  You need to get streamId and privateKey at data.sparkfun.com and paste them
 *  below. Or just customize this script to talk to other HTTP servers.
 *
 */

#include <WiFi.h>

const char* ssid     = "Xiaomi_8912";
const char* password = "M10s2I87kQpT";

const char* host = "data.sparkfun.com";
const char* streamId   = "....................";
const char* privateKey = "....................";

void setup()
{
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

int value = 0;

void loop()
{
    delay(5000);
    ++value;

    Serial.print("connecting to ");
    Serial.println(host);

    // Use WiFiClient class to create TCP connections
    WiFiClient client;
    const int httpPort = 80;
    if (!client.connect(host, httpPort)) {
        Serial.println("connection failed");
        return;
    }

    // We now create a URI for the request
    String url = "/input/";
    url += streamId;
    url += "?private_key=";
    url += privateKey;
    url += "&value=";
    url += value;

    Serial.print("Requesting URL: ");
    Serial.println(url);

    // This will send the request to the server
    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "Connection: close\r\n\r\n");
    unsigned long timeout = millis();
    while (client.available() == 0) {
        if (millis() - timeout > 5000) {
            Serial.println(">>> Client Timeout !");
            client.stop();
            return;
        }
    }

    // Read all the lines of the reply from server and print them to Serial
    while(client.available()) {
        String line = client.readStringUntil('\r');
        Serial.print(line);
    }

    Serial.println();
    Serial.println("closing connection");
}

```

and this is that I see in the serial port:
```

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3ffc0008,len:4
load:0x3ffc000c,len:1708
load:0x40078000,len:3124
load:0x40080000,len:256
entry 0x40080034
tcpip_task_hdlxxx : 3ffd5b70, prio:18,stack:2048
I (1445) wifi: frc2_timer_task_hdl:3ffd7ff8, prio:22, stack:2048
I (1456) wifi: pp_task_hdl : 3ffda7fc, prio:23, stack:8192


Connecting to Xiaomi_8912
I (1472) wifi: mode : softAP (24:0a:c4:00:3a:17)
I (1473) wifi: mode : sta (24:0a:c4:00:3a:16) + softAP (24:0a:c4:00:3a:17)
dhcp server start:(ip: 192.168.4.1, mask: 255.255.255.0, gw: 192.168.4.1)
....I (3679) wifi: reconnect
....I (5883) wifi: reconnect
.....I (8087) wifi: reconnect
....I (10291) wifi: reconnect
.....I (12494) wifi: reconnect
```

me-no-dev, 2017-02-01:
sketch looks totally fine, but the WiFi can not connect to your access point. Can you connect to the ESP's WiFi? it looks like it's starting it's AP

PeWos, 2017-02-01:
Sometimes, now i have found it and I have connect to the ESP

closed 2017-02-01 by me-no-dev

PeWos, 2017-02-02:
@me-no-dev Still does not work, when I load the program I can see the access point of the esp32 and then when I should change to the STA mode I do not see that connect, it is necessary to power the module through an external source instead of connecting it to the pc only?

me-no-dev, 2017-02-02:
please do paste some serial output

PeWos, 2017-02-02:
Is the same that yesterday:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3ffc0008,len:4
load:0x3ffc000c,len:1708
load:0x40078000,len:3124
load:0x40080000,len:256
entry 0x40080034
tcpip_task_hdlxxx : 3ffd5b70, prio:18,stack:2048
I (1445) wifi: frc2_timer_task_hdl:3ffd7ff8, prio:22, stack:2048
I (1456) wifi: pp_task_hdl : 3ffda7fc, prio:23, stack:8192
�

Connecting to Xiaomi_8912
I (1472) wifi: mode : softAP (24:0a:c4:00:3a:17)
I (1473) wifi: mode : sta (24:0a:c4:00:3a:16) + softAP (24:0a:c4:00:3a:17)
dhcp server start:(ip: 192.168.4.1, mask: 255.255.255.0, gw: 192.168.4.1)
....I (3679) wifi: reconnect
....I (5883) wifi: reconnect
.....I (8087) wifi: reconnect
....I (10291) wifi: reconnect
.....I (12494) wifi: reconnect

me-no-dev, 2017-02-02:
as I said yesterday, look for a problem with your access point. The esp seems to be OK but can not connect to your AP