espressif/arduino-esp32 #166: Make i2cSetFrequency Error

SirRibeiro, 2017-02-02:
i was trying to upgrade the arduino-esp32 component, and i get the following error when trying to execute make:
`
C:/esp32_projects/.../components/arduino/cores/esp32/esp32-hal-i2c.c: In function 'i2cSetFrequency':
C:/esp32_projects/.../components/arduino/cores/esp32/esp32-hal-i2c.c:350:29: error: 'volatile union <anonymous>' has no member named 'period'
     i2c->dev->scl_low_period.period = period;
                             ^
C:/esp32_projects/.../components/arduino/cores/esp32/esp32-hal-i2c.c: In function 'i2cGetFrequency':
C:/esp32_projects/.../components/arduino/cores/esp32/esp32-hal-i2c.c:378:50: error: 'volatile union <anonymous>' has no member named 'period'
     return APB_CLK_FREQ/(i2c->dev->scl_low_period.period+i2c->dev->scl_high_period.period);
                                                  ^
C:/esp32_projects/.../components/arduino/cores/esp32/esp32-hal-i2c.c:379:1: error: control reaches end of non-void function [-Werror=return-type]

cc1.exe: some warnings being treated as errors

`

I'm using as ESP-IDF component.

What i'm i doing wrong?

Thanks

me-no-dev, 2017-02-02:
you are using old IDF :)

SirRibeiro, 2017-02-02:
Indeed you are right of course :)

Thanks for the quick response

me-no-dev, 2017-02-02:
you're welcome :)

closed 2017-02-02 by me-no-dev