espressif/arduino-esp32 #167: WifiServer disconnects client.

feelfreelinux, 2017-02-02:
When I send message to client via WifiServer.avalible, the user is disconnected from server - I can connect to it again, so server is not stopping.
Sending the code - trying to write simple IRC bouncer.
`#include <WiFi.h>
WiFiClient client;
WiFiServer server(6667);
void setup()
{
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");

    WiFi.begin("ssid", "pass");

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    client.setNoDelay(true);
}

void loop()
{
  if(!client.connected()) {
    Serial.println("Connecting to irc");
    if(client.connect("chat.freenode.net", 6667)){
       Serial.println("connected, starting server");
       server.begin();
       delay(1000);
       client.println("USER esp 8 * : Esp User\r");
       delay(500);
       client.println("NICK esp32-test\r");
       delay(500);
       client.println("JOIN #esp32\r");
       delay(500);
       bool string = false;
       String line = "";
       while(true){
    if(client.available()) {
      char c = client.read();
      if(c==':')string = true;
      if(string)line+=c;
      if(c=='\n'){ // There we sent line to Serial.
       Serial.print(line);

       // handle PING message
       if(line.substring(0,4)=="PING"){
        line.replace("PING", "PONG");
        const char *response = line.c_str();
        client.write(response, sizeof(response));
        Serial.print(line);
       } else if (server.available()){
        Serial.println("Client is conencted.");
        WiFiClient BNC = server.available();
        if(BNC.connected()){
          Serial.println("Sending message to client...");
         const char *response = line.c_str();
         server.available().write(response, sizeof(response));
         delay(1);
        }
       }
       line=""; 
      }
     }
    }

     }
   }
}
`

Dennis650, 2017-02-02:
Is this https://github.com/espressif/arduino-esp32/issues/119 the same problem?

Greetings
Dennis

me-no-dev, 2017-02-02:
The biggest mistake is that you get the client with server.available, but then send to another server.available which might or might not be there. Send to BNC instead

feelfreelinux, 2017-02-03:
@me-no-dev The problem still occur. Hmm, yes looks related to this issue.

me-no-dev, 2017-03-02:
issue is resolved in the latest commits. I'm closing this now :)

closed 2017-03-02 by me-no-dev