espressif/arduino-esp32 #168: Unable to upload schetch

ionciubotaru, 2017-02-03:
I am using arduino IDE 1.8.1
I follow the instruction to download esp 32 from github and I run get.exe
I have a DEV board from olimex [https://www.olimex.com/Products/IoT/ESP32-CoreBoard/](https://www.olimex.com/Products/IoT/ESP32-CoreBoard/)
The board works on com30, I select ESP32 DEV Module, 
When I put the board in programming mode I received on serial port:

> ets Jun  8 2016 00:22:57
> rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
> waiting for download

When I hit the upload button I received:

> Sketch uses 385797 bytes (36%) of program storage space. Maximum is 1044464 bytes.
> Global variables use 30348 bytes (10%) of dynamic memory, leaving 264564 bytes for local variables. Maximum is 294912 bytes.
> esptool.py v2.0-dev
> Connecting...
> 
> A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('a')
> A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('a')

 Please help

me-no-dev, 2017-02-03:
try lower upload speed and hold the boot button while it's trying to upload the firmware

copercini, 2017-02-03:
compile your sketch, and press the ESP flash button after arduino say the message in white:
Sketch uses X bytes (Y%) of program storage space. Maximum is Z bytes

ionciubotaru, 2017-02-03:
Thank you, I succed keeping GPIO0 low during upload.

closed 2017-02-03 by ionciubotaru