espressif/arduino-esp32 #173: Autoconnect

plerup, 2017-02-05:
In ESP8266/Arduino you only have to call WiFi.begin with the ssid and password in one run of your application and you could then remove the call to WiFi.begin and the esp will still automatically connect to the last known access point in subsequent runs.

For esp32 this doesn't seems to work. I've tried WiFi.setAutoConnect(true) but to no avail although subsequent calls to WiFi.getAutoConnect() returns 1

Any ideas?

me-no-dev, 2017-02-05:
yes, though not the case yet, stack will be selectable at any point, which means that wifi does not start before you call WiFi.begin. There is an option to autostart, but you have to use Arduino as IDF component.

plerup, 2017-02-05:
Ok, I tried with WiFi.begin() without parameters but that didn't work either

me-no-dev, 2017-02-05:
It started fine for me. I have just WiFi.begin();
```
[D][WiFiGeneric.cpp:174] _eventCallback(): 2 - STA_START
[D][WiFiGeneric.cpp:174] _eventCallback(): 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:174] _eventCallback(): 7 - STA_GOT_IP
STA Connected
STA SSID: nbis-test
STA IPv4: 192.168.254.199
```

plerup, 2017-02-05:
How do you enable this debug output?

me-no-dev, 2017-02-05:
add [here](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal.h#L48)
```cpp
#undef CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL
#define CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL ARDUHAL_LOG_LEVEL_DEBUG
```

me-no-dev, 2017-02-05:
actually [here](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-log.h#L23)

me-no-dev, 2017-02-05:
it's defined in tools/sdk/include/config/sdkconfig.h

plerup, 2017-02-05:
When using WiFi.begin(ssid, pw) I get this

    [D][WiFiGeneric.cpp:171] _eventCallback(): 12
    [D][WiFiGeneric.cpp:171] _eventCallback(): 2
    [D][WiFiGeneric.cpp:171] _eventCallback(): 4
    [D][WiFiGeneric.cpp:171] _eventCallback(): 7

When using WiFiBegin() I only get this

    [D][WiFiGeneric.cpp:171] _eventCallback(): 12
    [D][WiFiGeneric.cpp:171] _eventCallback(): 2

me-no-dev, 2017-02-05:
interesting why is your AP starting also. could you please pull the latest commits and switch to branch "me-no-dev" then try again?

me-no-dev, 2017-02-05:
@plerup if you are online why not come to gitter :) will speed things up

plerup, 2017-02-06:
Still can not get this to work, also with latest commit. Simple test:

    #include <WiFi.h>
    void setup() {
      Serial.begin(115200);
      Serial.println("Started");
      //WiFi.begin("xyz", "xyz");
      WiFi.begin();
      while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
      }
      Serial.println("\nWiFi connected"); 
    }
    void loop() {
    }

plerup, 2017-02-08:
Adding:

    WiFi.mode(WIFI_STA);

Gives an initial:

    [D][WiFiGeneric.cpp:176] _eventCallback(): Event: 12 - AP_START
    [D][WiFiGeneric.cpp:176] _eventCallback(): Event: 13 - AP_STOP
    [D][WiFiGeneric.cpp:176] _eventCallback(): Event: 2 - STA_START

otherwise still the same, no connection unless ssid pw

Tried this on two different type of boards now

me-no-dev, 2017-02-08:
no idea :) but I'll research some ideas I have, that still do not explain why it works on my end... did I do something in the past or what? no idea really.

me-no-dev, 2017-02-08:
@plerup change [this](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.cpp#L62) to ```WIFI_STORAGE_FLASH``` and see if it helps. The fact that it works for me might be a bug :)

plerup, 2017-02-09:
Cool! Will try this tonight

plerup, 2017-02-09:
@me-no-dev 

Yes!
Now it works, took away the AP_START as well. 

Great :+1:

me-no-dev, 2017-02-09:
awesome. I want to finalize the partition scheme and will commit the change :)

closed 2017-02-10 by me-no-dev

auladinamica, 2017-12-13:
hi guys i receive this error when try to create ap modein this line 👍 

WiFi.mode(WIFI_AP_STA);

error :  
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 13 - AP_STOP

whats is wrong ?

thanks

tablatronix, 2018-01-24:
Why was this closed if it is not resolved ?
autoconnect does not work as far as I can tell, you still must call WiFi.begin().

tablatronix, 2018-01-24:
ahh it looks like mode is not persistent, it works if you 
WiFi.mode(WIFI_STA); on setup()

Is this expected?

tablatronix, 2018-02-19:
@me-no-dev bump, since this is not yet implemented, it should remain an open issue please.

tablatronix, 2018-07-20:
deprecated
https://github.com/espressif/esp-idf/issues/1960