espressif/arduino-esp32 #176: esp32-snippets

tobozo, 2017-02-06:
Found this repo : https://github.com/nkolban/esp32-snippets 

How far is it from being usable in the Arduino IDE ?

me-no-dev, 2017-02-06:
all of it is available. Everything in IDF is available here. Most of the things I saw as peripherals and network are already implemented in Arduino language. Exception still are PWM, RMT, PCNT and I2S, but again, those are available through the included idf libs.
@nkolban could tell better what are the examples based on.

tobozo, 2017-02-07:
Thanks, closing this as it was more about where to find ESP32 snippets in the Arduino IDE and just realized the URL I pasted was from another repo :man_facepalming:

closed 2017-02-07 by tobozo

rdheiliger, 2017-04-13:
I am trying to use PCNT in this program. I am not getting any counts when i read the counter value. The program compiles and i get back ESP_OK for the configuration and the read counter value lines. I am using a  pulse generator to input pulses on pins 27 and 14. Any thoughts?

I tried using interrupts to count pulses on a pin, but interrupts can only handle 3 to 4 pulses/second and inputs 27 and 14 work with interrupts.

void initPcnt(){

//initialize counter for flow0
pcnt_config_t pcnt_config =  {
    27,             /*!< Pulse input gpio_num, if you want to use gpio16, pulse_gpio_num = 16, a negative value will be ignored */
    PCNT_PIN_NOT_USED,              /*!< Contol signal input gpio_num, a negative value will be ignored*/
    PCNT_MODE_KEEP,    /*!< PCNT low control mode*/
    PCNT_MODE_KEEP,    /*!< PCNT high control mode*/
    PCNT_COUNT_INC,     /*!< PCNT positive edge count mode*/
    PCNT_COUNT_DIS,     /*!< PCNT negative edge count mode*/
    32767,          /*!< Maximum counter value */
    0,         /*!< Minimum counter value */
    PCNT_UNIT_0,               /*!< PCNT unit number */
    PCNT_CHANNEL_0         /*!< the PCNT channel */
};
  if(pcnt_unit_config(&pcnt_config) == ESP_OK)        //init unit
      Serial.println("Config Unit_0 = ESP_OK");

  pcnt_set_filter_value(PCNT_UNIT_0, 1);    
  pcnt_filter_enable(PCNT_UNIT_0);
    
pcnt_config_t pcnt_config2 = {
   14,             /*!< Pulse input gpio_num, if you want to use gpio16, pulse_gpio_num = 16, a negative value will be ignored */
    PCNT_PIN_NOT_USED,              /*!< Contol signal input gpio_num, a negative value will be ignored*/
    PCNT_MODE_KEEP,    /*!< PCNT low control mode*/
    PCNT_MODE_KEEP,    /*!< PCNT high control mode*/
    PCNT_COUNT_INC,     /*!< PCNT positive edge count mode*/
    PCNT_COUNT_DIS,     /*!< PCNT negative edge count mode*/
    32767,          /*!< Maximum counter value */
    0,         /*!< Minimum counter value */
    PCNT_UNIT_1,               /*!< PCNT unit number */
    PCNT_CHANNEL_0,         /*!< the PCNT channel */
 };
  if(pcnt_unit_config(&pcnt_config2) == ESP_OK)        //init unit
    Serial.println("Config Unit_1 = ESP_OK");

  pcnt_set_filter_value(PCNT_UNIT_1, 1);    
  pcnt_filter_enable(PCNT_UNIT_1);
    
}

void readPcntCounter_0(){
  
  if(pcnt_get_counter_value(PCNT_UNIT_0, &flow0Counter) == ESP_OK)
    Serial.println("Read pcnt unit 0");
      delay(10);
    Serial.print("flow0Counts = ");
    Serial.println(flow0Counter); 

}

void readPcntCounter_1(){
  
   if(pcnt_get_counter_value(PCNT_UNIT_1, &flow1Counter) == ESP_OK)
      Serial.println("Read pcnt unit 1");
      delay(10);
    Serial.print("flow1Counts = ");
    Serial.println(flow1Counter); 

  
}

onpagithub, 2017-08-29:
rdheiliger: Any solution of problem with Pulse Counter? I have the same result.... No reaction for pulses....

ToninoTarsi, 2018-02-16:
Same problem here !. Anyone got PCNT working with Arduino IDE ?
Thanks

onpagithub, 2018-02-16:
PCNT in Arduino IDE (PlatformIO) is working for me. I'm using it as Frequency Counter.

ToninoTarsi, 2018-02-16:
Thanks.. And how did you solve the code obove?

onpagithub, 2018-02-16:
Honestly, I played with the code very long time and read a lot of on the internet discussions. Now, the code is part of huge project, I can't remove it form it, don't know, what exactly must be preset or whatever.

Here is my code. Timer counts 1s, during this time are counted pulses comming to pin.

/*
  Příklad pro měření kmitočtu pomocí Pulse Counter hardware
  LED dioda bliká jako identifikace hradla / gate - doba, kdy se měří pulsy

  Ověřeno - funguje

*/
extern "C" {
  #include "soc/pcnt_struct.h"
}
#include "driver/pcnt.h"

/* LED pin */
byte ledPin = 0;
/* otevření relé */
byte relayPin = 14;
/* vstupní pin pro měření kmitočtu */
byte pulsePin = 34;
/* counter - proměnná, kde čteme načtené pulsy */
int16_t flow0Counter = 0;
int16_t Pulses = 0;
int16_t x;
volatile uint32_t us_time, us_time_diff;

/* pro timer (dobu), během kterého se budou měřit pulsy */
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

volatile byte state = LOW;
volatile byte state2 = LOW;
volatile byte state_tmr = 0;
volatile byte value_ready = 0;

#define PCNT_TEST_UNIT     PCNT_UNIT_0
#define PCNT_H_LIM_VAL     32767
#define PCNT_L_LIM_VAL     -1

void setup() {
  /* LED - výstup pro LED */
  pinMode(ledPin, OUTPUT);
  pinMode(relayPin, OUTPUT);
  digitalWrite(relayPin, HIGH);
  /* vstupní pin */
  pinMode(pulsePin,INPUT);
  Serial.begin(115200);
  Serial.println("");
  initPcnt();
  /* nastavíme timer a alarm
    - Use 1st timer of 4 (counted from zero).
    - Set 80 divider for prescaler (see ESP32 Technical Reference Manual for more info
  */
  timer = timerBegin(0, 80, true);
  /* Attach onTimer function to our timer. */
  timerAttachInterrupt(timer, &onTimer, true);
  // Set alarm to call onTimer function every second (value in microseconds).
  // Repeat the alarm (third parameter)
  timerAlarmWrite(timer, 1000000, true);
  // Start an alarm
  timerAlarmEnable(timer);
}

void loop() {
  Serial.println("Smycka");
  readPcntCounter_0();
  if(value_ready == 1) {
    Serial.printf("Nacteno: %d\r\n",Pulses);
    Serial.printf("Frekvence %f Hz\r\n",(float)Pulses);
  }
  delay(100);
}

/* obluha alarmu timeru */
void IRAM_ATTR onTimer(){
  portENTER_CRITICAL_ISR(&timerMux);
  switch (state_tmr) {

    case 0 :
      /* spusíme počítání pulsů */
      pcnt_counter_clear(PCNT_TEST_UNIT);
      pcnt_counter_resume(PCNT_TEST_UNIT);
      digitalWrite(ledPin, HIGH);
      value_ready = 0;
      state_tmr = 1;
      break;

    case 1 :
      /* ukončíme počítání pulsů */
      pcnt_counter_pause(PCNT_TEST_UNIT);
      pcnt_get_counter_value(PCNT_TEST_UNIT, &flow0Counter);
      Pulses = flow0Counter;
      flow0Counter = 0;
      digitalWrite(ledPin, LOW);
      value_ready = 1;
      state_tmr = 2;
      break;

    case 2 :
      /* počkáme vteřinu - vynecháme jedno volání timeru */
      state_tmr = 0;
      break;

    default:
      break;
  }
  portEXIT_CRITICAL_ISR(&timerMux);
}

//initialize counter for flow0
void initPcnt() {
  Serial.println("Inicializace pulse totalizeru");

  pcnt_config_t pcnt_config = {
    pulsePin, // Pulse input gpio_num, if you want to use gpio16, pulse_gpio_num = 16, a negative value will be ignored
    PCNT_PIN_NOT_USED, // Control signal input gpio_num, a negative value will be ignored
    PCNT_MODE_KEEP, // PCNT low control mode
    PCNT_MODE_KEEP, // PCNT high control mode
    PCNT_COUNT_INC, // PCNT positive edge count mode
    PCNT_COUNT_DIS, // PCNT negative edge count mode
    PCNT_H_LIM_VAL, // Maximum counter value
    PCNT_L_LIM_VAL, // Minimum counter value
    PCNT_TEST_UNIT, // PCNT unit number
    PCNT_CHANNEL_0, // the PCNT channel
  };

  if(pcnt_unit_config(&pcnt_config) == ESP_OK) //init unit
    Serial.println("Config Unit_0 = ESP_OK");

  /* nastavení hodnoty pro filtrování */
  pcnt_set_filter_value(PCNT_TEST_UNIT, 1);
  /* povolíme vstupní filtr */
  pcnt_filter_enable(PCNT_TEST_UNIT);

  /* Nastavíme všechny eventy */
  pcnt_intr_disable(PCNT_TEST_UNIT);
  pcnt_event_disable(PCNT_TEST_UNIT, PCNT_EVT_L_LIM);
  pcnt_event_disable(PCNT_TEST_UNIT, PCNT_EVT_H_LIM);
  pcnt_event_disable(PCNT_TEST_UNIT, PCNT_EVT_THRES_0);
  pcnt_event_disable(PCNT_TEST_UNIT, PCNT_EVT_THRES_1);
  pcnt_event_disable(PCNT_TEST_UNIT, PCNT_EVT_ZERO);

  /*Pause counter*/
  pcnt_counter_pause(PCNT_TEST_UNIT);
  /*Reset counter value*/
  pcnt_counter_clear(PCNT_TEST_UNIT);
  /* povolíme počítání pulsů */
  pcnt_intr_enable(PCNT_TEST_UNIT);
  /*Resume counting*/
  pcnt_counter_resume(PCNT_TEST_UNIT);

}

void readPcntCounter_0() {
  if(pcnt_get_counter_value(PCNT_TEST_UNIT, &flow0Counter) == ESP_OK)
  Serial.println("Read pcnt unit 0");
  delay(10);
  Serial.print("flow0Counts = ");
  Serial.println(flow0Counter);
}

ToninoTarsi, 2018-02-16:
Thanks for your time :-)

jumpjack, 2019-08-21:
Formatted code:
`
/*
Příklad pro měření kmitočtu pomocí Pulse Counter hardware
LED dioda bliká jako identifikace hradla / gate - doba, kdy se měří pulsy

Ověřeno - funguje

*/
extern "C" {
#include "soc/pcnt_struct.h"
}
#include "driver/pcnt.h"

/* LED pin /
byte ledPin = 0;
/ otevření relé /
byte relayPin = 14;
/ vstupní pin pro měření kmitočtu /
byte pulsePin = 34;
/ counter - proměnná, kde čteme načtené pulsy */
int16_t flow0Counter = 0;
int16_t Pulses = 0;
int16_t x;
volatile uint32_t us_time, us_time_diff;

/* pro timer (dobu), během kterého se budou měřit pulsy */
hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

volatile byte state = LOW;
volatile byte state2 = LOW;
volatile byte state_tmr = 0;
volatile byte value_ready = 0;

#define PCNT_TEST_UNIT PCNT_UNIT_0
#define PCNT_H_LIM_VAL 32767
#define PCNT_L_LIM_VAL -1

void setup() {
/* LED - výstup pro LED /
pinMode(ledPin, OUTPUT);
pinMode(relayPin, OUTPUT);
digitalWrite(relayPin, HIGH);
/ vstupní pin /
pinMode(pulsePin,INPUT);
Serial.begin(115200);
Serial.println("");
initPcnt();
/ nastavíme timer a alarm
- Use 1st timer of 4 (counted from zero).
- Set 80 divider for prescaler (see ESP32 Technical Reference Manual for more info
/
timer = timerBegin(0, 80, true);
/ Attach onTimer function to our timer. */
timerAttachInterrupt(timer, &onTimer, true);
// Set alarm to call onTimer function every second (value in microseconds).
// Repeat the alarm (third parameter)
timerAlarmWrite(timer, 1000000, true);
// Start an alarm
timerAlarmEnable(timer);
}

void loop() {
Serial.println("Smycka");
readPcntCounter_0();
if(value_ready == 1) {
Serial.printf("Nacteno: %d\r\n",Pulses);
Serial.printf("Frekvence %f Hz\r\n",(float)Pulses);
}
delay(100);
}

/* obluha alarmu timeru */
void IRAM_ATTR onTimer(){
portENTER_CRITICAL_ISR(&timerMux);
switch (state_tmr) {

case 0 :
  /* spusíme počítání pulsů */
  pcnt_counter_clear(PCNT_TEST_UNIT);
  pcnt_counter_resume(PCNT_TEST_UNIT);
  digitalWrite(ledPin, HIGH);
  value_ready = 0;
  state_tmr = 1;
  break;

case 1 :
  /* ukončíme počítání pulsů */
  pcnt_counter_pause(PCNT_TEST_UNIT);
  pcnt_get_counter_value(PCNT_TEST_UNIT, &flow0Counter);
  Pulses = flow0Counter;
  flow0Counter = 0;
  digitalWrite(ledPin, LOW);
  value_ready = 1;
  state_tmr = 2;
  break;

case 2 :
  /* počkáme vteřinu - vynecháme jedno volání timeru */
  state_tmr = 0;
  break;

default:
  break;
}
portEXIT_CRITICAL_ISR(&timerMux);
}

//initialize counter for flow0
void initPcnt() {
Serial.println("Inicializace pulse totalizeru");

pcnt_config_t pcnt_config = {
pulsePin, // Pulse input gpio_num, if you want to use gpio16, pulse_gpio_num = 16, a negative value will be ignored
PCNT_PIN_NOT_USED, // Control signal input gpio_num, a negative value will be ignored
PCNT_MODE_KEEP, // PCNT low control mode
PCNT_MODE_KEEP, // PCNT high control mode
PCNT_COUNT_INC, // PCNT positive edge count mode
PCNT_COUNT_DIS, // PCNT negative edge count mode
PCNT_H_LIM_VAL, // Maximum counter value
PCNT_L_LIM_VAL, // Minimum counter value
PCNT_TEST_UNIT, // PCNT unit number
PCNT_CHANNEL_0, // the PCNT channel
};

if(pcnt_unit_config(&pcnt_config) == ESP_OK) //init unit
Serial.println("Config Unit_0 = ESP_OK");

/* nastavení hodnoty pro filtrování /
pcnt_set_filter_value(PCNT_TEST_UNIT, 1);
/ povolíme vstupní filtr */
pcnt_filter_enable(PCNT_TEST_UNIT);

/* Nastavíme všechny eventy */
pcnt_intr_disable(PCNT_TEST_UNIT);
pcnt_event_disable(PCNT_TEST_UNIT, PCNT_EVT_L_LIM);
pcnt_event_disable(PCNT_TEST_UNIT, PCNT_EVT_H_LIM);
pcnt_event_disable(PCNT_TEST_UNIT, PCNT_EVT_THRES_0);
pcnt_event_disable(PCNT_TEST_UNIT, PCNT_EVT_THRES_1);
pcnt_event_disable(PCNT_TEST_UNIT, PCNT_EVT_ZERO);

/Pause counter/
pcnt_counter_pause(PCNT_TEST_UNIT);
/Reset counter value/
pcnt_counter_clear(PCNT_TEST_UNIT);
/* povolíme počítání pulsů */
pcnt_intr_enable(PCNT_TEST_UNIT);
/Resume counting/
pcnt_counter_resume(PCNT_TEST_UNIT);

}

void readPcntCounter_0() {
if(pcnt_get_counter_value(PCNT_TEST_UNIT, &flow0Counter) == ESP_OK)
Serial.println("Read pcnt unit 0");
delay(10);
Serial.print("flow0Counts = ");
Serial.println(flow0Counter);
}
`

jumpjack, 2019-08-21:
---