espressif/arduino-esp32 #179: Using Arduino code on FreeRTOS

seopyoon, 2017-02-07:
Hello,

I want to implement something very similar to `SimpleWiFiServer` on WiFi example from this repo.  I want to add that to my current project written in C using esp-idf template.  

How should I go about using the two together??  Should I port my C code to Arduino somehow?? Or, port Arduino code over to esp-idf??  

In the esp-idf, there doesn't exist WiFi Client and etc., so I do not know how to go about merging two projects written in two different environments.

me-no-dev, 2017-02-07:
Have you looked in the readme of the project? It has instructions on how to run arduino together with IDF (Arduino being IDF component). That will give you what you ask :)

seopyoon, 2017-02-07:
I apologise for not having read your README thoroughly.  
Works great!  

Mine is already a big application, and wonder if including Arduino library will somehow hit the limit?  Will try to merge tomorrow.

me-no-dev, 2017-02-07:
you will hit the limit based on your code, not on the fact that you have the arduino component there :) code is not included if not used

closed 2017-02-07 by me-no-dev