espressif/arduino-esp32 #180: lwip_connect_r:113

PeWos, 2017-02-07:
hi, i have a problem with the wificlient, sometimes the serial port show me 
`[E][WiFiClient.cpp:74]` connect(): lwip_connect_r: `113`

I am doing with the example wificlient:

```
/*
 *  This sketch sends data via HTTP GET requests to data.sparkfun.com service.
 *
 *  You need to get streamId and privateKey at data.sparkfun.com and paste them
 *  below. Or just customize this script to talk to other HTTP servers.
 *
 */

#include <WiFi.h>

const char* ssid     = "your-ssid";
const char* password = "your-password";

const char* host = "data.sparkfun.com";
const char* streamId   = "....................";
const char* privateKey = "....................";

void setup()
{
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

int value = 0;

void loop()
{
    delay(5000);
    ++value;

    Serial.print("connecting to ");
    Serial.println(host);

    // Use WiFiClient class to create TCP connections
    WiFiClient client;
    const int httpPort = 80;
    if (!client.connect(host, httpPort)) {
        Serial.println("connection failed");
        return;
    }

    // We now create a URI for the request
    String url = "/input/";
    url += streamId;
    url += "?private_key=";
    url += privateKey;
    url += "&value=";
    url += value;

    Serial.print("Requesting URL: ");
    Serial.println(url);

    // This will send the request to the server
    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "Connection: close\r\n\r\n");
    unsigned long timeout = millis();
    while (client.available() == 0) {
        if (millis() - timeout > 5000) {
            Serial.println(">>> Client Timeout !");
            client.stop();
            return;
        }
    }

    // Read all the lines of the reply from server and print them to Serial
    while(client.available()) {
        String line = client.readStringUntil('\r');
        Serial.print(line);
    }

    Serial.println();
    Serial.println("closing connection");
}

```

me-no-dev, 2017-02-08:
Code is fine, error 113 is ECONNABORTED (Connection aborted). Try flushing the client or reading all available data. ```client.readStringUntil('\r');``` does not read it all

PeWos, 2017-02-08:
Thank you now it works fine, I have another problem: how can I declare an attachinterrupt of a sensor in pin 33?
Thanks in advance

me-no-dev, 2017-02-08:
the same way as any other pin in Arduino :) attachInterrupt(33, method, direction);

closed 2017-02-08 by me-no-dev

bioyavuz, 2018-01-14:
I had the same problem. After spending few days, a minute detail took my attention.  I think I solved my problem in an interesting way:

Whenever, I use tools "tools->Core Debug Level" other than "None" I had this Error:
[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 118

**Changing "Core Debug Level" to "None" solved my problem.**

Can somebody explain why?

robertcasanova, 2018-04-24:
I think this issue is somehow related also to Internet Connection. I spent all afternoon trying to figure out how to fix it and then I retry with an Hotspot from my mobile phone and it works perfect.

robertcasanova, 2018-04-27:
This issue is still present.
The connection seems very unstable and sometimes works well and some others i have this error:

`[E][WiFiClient.cpp:97] connect(): lwip_connect_r: 113`

I'm using a Adafruit Huzzah32 but should not be board related.
To check this behaviour I'm using the Example sketch: Wifi -> WifiClient without modifications.

Any ideas? At least if there's a workaround because at the moment Wifi connection is not usable at all for a project.

Thx for help!

cybergogo, 2018-10-10:
up? same issue here