espressif/arduino-esp32 #181: Low power mode

plerup, 2017-02-07:
Is it possible in any way to enter a more low power mode here.

By using
    esp_wifi_set_ps(1)
in the IDF you can significantly reduce the power consumption. But here in Arduino this doesn't have any effect.

I guess that the Arduino loop context prevents low power but if I put a delay in the loop I would have expected some reduction, or am I completely wrong here?

me-no-dev, 2017-02-07:
this is strange to say the least. arduino includes all IDF components and you are talking about wifi power save, which has nothing to do with execution loop :) By all means it should have worked and I'll give it a go here

plerup, 2017-02-08:
My thought was that the constant execution in the loop meant that the CPU was running at full speed and hence draw current.

me-no-dev, 2017-02-08:
we are using the latest SDK as of 2 days ago. I also tried to change ps while measuring and can confirm that it does not change, even though I am running as IDF component.
Here is another though: you are calling it from app_main or from task running on core 0, and there is a bug that prevents it from working on core 1. Can you please test and confirm that? Maybe share some IDF code that you have working?

plerup, 2017-02-08:
Well I just took the example at esp-idf/examples/protocols/http_request/ and added esp_wifi_set_ps(1)

Is that what you did as well?

me-no-dev, 2017-02-08:
no I tried running it from Arduino loop :) and yes the example code does run from app_main (I imagine you added the line there or in ```initialise_wifi();```, though you did not specify). So that probably is the cause. I have already submitted some reports to the colleagues, so I hope it will be resolved soon.
There is more than that, that does not work from core 1 (where Arduino runs)

plerup, 2017-02-08:
In the http_get_task procedure 

        xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT,
                            false, true, portMAX_DELAY);
        ESP_LOGI(TAG, "Connected to AP");
 
        int ps_mode = 1;
        int err = esp_wifi_set_ps(ps_mode);
 
        ESP_LOGI(TAG, "Enter Power mode %d, res=%d", ps_mode, err);

me-no-dev, 2017-02-08:
so, let's recap.
1. It's working in an IDF example (```http_get_task``` also runs on core 0)
2. It's not working in Arduino

Reason: Arduino runs all of it's code (except WiFi and PHY initialization) on Core 1, while IDF examples run all code (yours and system) on core 0. Everything is working fine on core ) but is not yet working correctly on core 1 (that is why Arduino initializes WiFi and phy in app_main on core 0)

Resolution: I have submitted bug reports to Espressif and I am awaiting responses. When WiFi is fixed to run properly on core 1, this problem will be fixed as well.

plerup, 2017-02-08:
:+1: Thanks a lot for your support!

Just an ignorant wondering, why the difference in IDF and Arduino, i.e. why can't Arduino run on core 0?

me-no-dev, 2017-02-08:
it can run on core 0 but then nothing will be running on core 1 :) so Arduino and the network and all of the rest running on core 0 would have to share the CPU time

me-no-dev, 2017-02-08:
on a side note, it's great that Arduino runs on core 1, because nothing else does, so we get to find bugs like this :)

plerup, 2017-02-09:
Tried running from core 0 now, no difference in power consumption

me-no-dev, 2017-02-23:
@plerup some new wifi stuff got merged. Can you please retest this?

plerup, 2017-02-23:
Yes, the current consumption is different now depending on if esp_wifi_set_ps is called or not.

closed 2017-02-23 by plerup

me-no-dev, 2017-02-23:
great :) you can start/stop wifi and BT at will now :) check the added example to the WiFi lib

plerup, 2017-02-24:
Really nice! :+1: :smile: