espressif/arduino-esp32 #182: SoftAP Server issue

seopyoon, 2017-02-08:
Hi, I am trying to build an application, where user will pass desired SSID and its password by connecting to ESP32 board acting as an AP.  Then remember the necessary information for network connectivity.

So, I have the following [code](https://gist.github.com/seopyoon/6d5b15e7d479bb9cd12c65b3e660ad6e).  This code is not fully implemented to get SSID and password, but it should at least connect, and when accessed at `http://192.168.4.1` it should show `hello world`.

But, instead, I get the following terminal message
[image]

and, on phone, it cannot connect to the AP on the settings page.  I wanted something similar to [this](https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/example-sketch-ap-web-server).  `join` followed by an immediate `leave`

Could someone help me figure out where I might be doing wrong?

me-no-dev, 2017-02-08:
what is the phone telling you? from the above output is visible connect/disconnect of a device. Does it get IP? what is actually happening?

seopyoon, 2017-02-08:
The phone doesnt give any message. It attempts to connect and afrer about half a second of trying, it does not try any more. I press again, then it tries again for an even shorter time, and stops.

me-no-dev, 2017-02-08:
that is not much to go by now is it :) try computer or something. You need to have better idea why you get disconnected from the AP.

me-no-dev, 2017-02-08:
there are no errors in the logs above, code also looks fine (though you are running only on one core)

clowrey, 2017-02-08:
Some phones will disconnect from APs if they can't get internet access in a short period of time, not sure if this is that issue..

me-no-dev, 2017-02-08:
@clowrey good point! I had to do some tricks on mine too

seopyoon, 2017-02-08:
What kind of tricks?? 
The same happens on my mac. I dunno how I could get some message from wifi connect attempts on Mac.

me-no-dev, 2017-02-08:
try opening the Console.app and see if it says anything. As for the tricks, I had to confirm something somewhere... was pretty obscure (search in google).

plerup, 2017-02-08:
To handle this you need a "captive portal" which means you have to have a local DNS server which redirects everything to a server which provides a 200 http request answer

seopyoon, 2017-02-09:
@plerup you suggest that I implement something like [this](https://github.com/esp8266/Arduino/blob/master/libraries/DNSServer/examples/CaptivePortal/CaptivePortal.ino) in order to achieve what I am planning on doing?  (A SoftAP Server at initial launch to get information about home WiFi and password?)

plerup, 2017-02-09:
Yes, we are just missing the webserver for now...

seopyoon, 2017-02-13:
@plerup @me-no-dev Do we know when webserver will be available??

closed 2017-02-20 by seopyoon

eljefedelrodeodeljefe, 2017-04-22:
Has here been progress?