espressif/arduino-esp32 #187: Guru Meditation Error WiFi Server, new client

isgallagher, 2017-02-09:
Hello,

This is the message I'm getting occasionally after so many client connects/disconnects.  It happens only after a new client connects, but it doesn't always happen.

```
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x40085157  PS      : 0x00060a33  A0      : 0x80084634  A1      : 0x3ffdc650  
A2      : 0x0000001d  A3      : 0x3ffdc6d4  A4      : 0x00000004  A5      : 0x3ffcb7e8  
A6      : 0x3ffdc700  A7      : 0x00000000  A8      : 0x80082e80  A9      : 0x3ffdc640  
A10     : 0x00000003  A11     : 0x00060a23  A12     : 0x00060a23  A13     : 0x00000000  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002d  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x40085157:0x3ffdc650 0x40084634:0x3ffdc670 0x40083223:0x3ffdc690 0x400e5069:0x3ffdc6d0 0x400dc7bd:0x3ffdc700 0x400e52cc:0x3ffdc730 0x400e5750:0x3ffdc750 0x400dbaf5:0x3ffdc790 0x400dc0d0:0x3ffdc810 0x400dc141:0x3ffdc840 0x400d140a:0x3ffdc860 0x4010d051:0x3ffdc880 0x400d10d4:0x3ffdc8b0 0x40105eb5:0x3ffdc910

CPU halted.
```

me-no-dev, 2017-02-09:
duplicate :) see https://github.com/espressif/arduino-esp32/issues/109

closed 2017-02-09 by me-no-dev

isgallagher, 2017-02-09:
Guess I should have searched, my bad!