espressif/arduino-esp32 #188: IDF Integration Pending IDE Support

WALLTECH, 2017-02-09:
I would very much like to use BLE in the Arduino IDE and it is taking quite a while, but I was told that the IDF functionality could still be called in the IDE even though it isn't wrapped in a library yet. How would I go about doing this in the meantime, and also when will BLE be supported :)

WALLTECH, 2017-02-16:
Any help would be appreciated, this is an often asked question with no answer

me-no-dev, 2017-02-16:
It have actually been answered many many times. You can search through the issues and see. Opening another issue would not change a thing in the situation, if anything, it's annoying. The whole stack is under heavy work (not here or in idf, in out company work repo) and I have a few classes already written and few things that I am testing, but nothing is solid enough to release it.

closed 2017-02-16 by me-no-dev

WALLTECH, 2017-02-17:
Alright, thank you for your response and good luck. Thank you for your hard work