espressif/arduino-esp32 #189: Compilation Error: 'reason2str' not declared in this scope

forthlightning, 2017-02-09:
Running arduino as a component on mac OSX.

```
CC cores/esp32/libb64/cencode.o
CC cores/esp32/esp32-hal-adc.o
CC cores/esp32/esp32-hal-dac.o
CC cores/esp32/esp32-hal-gpio.o
CC cores/esp32/esp32-hal-i2c.o
CC cores/esp32/esp32-hal-ledc.o
CC cores/esp32/esp32-hal-matrix.o
CC cores/esp32/esp32-hal-misc.o
CC cores/esp32/esp32-hal-sigmadelta.o
CC cores/esp32/esp32-hal-spi.o
CC cores/esp32/esp32-hal-time.o
CC cores/esp32/esp32-hal-timer.o
CC cores/esp32/esp32-hal-touch.o
CC cores/esp32/esp32-hal-uart.o
CC cores/esp32/stdlib_noniso.o
CC cores/esp32/wiring_pulse.o
CC cores/esp32/wiring_shift.o
CXX cores/esp32/Esp.o
CXX cores/esp32/HardwareSerial.o
CXX cores/esp32/IPAddress.o
CXX cores/esp32/IPv6Address.o
CXX cores/esp32/MD5Builder.o
CXX cores/esp32/Print.o
CXX cores/esp32/Stream.o
CXX cores/esp32/StreamString.o
CXX cores/esp32/WMath.o
CXX cores/esp32/WString.o
CXX cores/esp32/base64.o
CXX cores/esp32/cbuf.o
CXX cores/esp32/main.o
CXX libraries/WiFi/src/WiFi.o
CXX libraries/WiFi/src/WiFiAP.o
CXX libraries/WiFi/src/WiFiClient.o
CXX libraries/WiFi/src/WiFiGeneric.o
In file included from .../components/arduino/cores/esp32/esp32-hal.h:49:0,
                 from .../components/arduino/cores/esp32/Arduino.h:39,
                 from .../components/arduino/libraries/WiFi/src/WiFiClient.h:24,
                 from .../components/arduino/libraries/WiFi/src/WiFi.h:37,
                 from .../components/arduino/libraries/WiFi/src/WiFiGeneric.cpp:25:
.../components/arduino/libraries/WiFi/src/WiFiGeneric.cpp: In static member function 'static esp_err_t WiFiGenericClass::_eventCallback(void*, system_event_t*)':
.../components/arduino/libraries/WiFi/src/WiFiGeneric.cpp:182:59: error: 'reason2str' was not declared in this scope
         log_w("Reason: %u - %s", reason, reason2str(reason));
                                                           ^
.../components/arduino/cores/esp32/esp32-hal-log.h:99:72: note: in definition of macro 'log_w'
 #define log_w(format, ...) log_printf(ARDUHAL_LOG_FORMAT(W, format), ##__VA_ARGS__)
                                                                        ^
make[1]: *** [libraries/WiFi/src/WiFiGeneric.o] Error 1
make: *** [arduino-build] Error 2
```

forthlightning, 2017-02-09:
Going in to 'make menuconfig' and then configuring the arduino component to have no debug output allows flashing to continue as expected.

me-no-dev, 2017-02-10:
Aaah :) good catch! Fixed https://github.com/espressif/arduino-esp32/commit/0101564219839f3159c954d8d0d40b2ab0917739

closed 2017-02-10 by me-no-dev