espressif/arduino-esp32 #190: Is there Bluetooth example for arduino?

pwreng, 2017-02-10:
Hi, I have ESP WROOM 32 (support Wifi&BLE). I have been looking for an Arduino example using BLE.
I have one for Eclipse which seems too much complicated for me.

Any help is appreciated.

James

PeWos, 2017-02-10:
As far as I know, BT is not implemented in arduino yet

me-no-dev, 2017-02-10:
You can use everything available through IDF. I see lots of work still being done on the BT stack and I see no point into investing lots of time into something that is not finalized and instead improve and integrate things that are done.
Surely there will be Arduino "integration" in the means of easier-to-handle classes :)
Imagine what will happen now with issues here if something in BT breaks in a mean time. People will blame Arduino and I'll be chasing my tail :D

p.s. Please try search issue before submitting a duplicate issue. I'm closing this as there are at least 2 others available

closed 2017-02-10 by me-no-dev

andrei-ivanov, 2017-02-10:
Well, it's good that we at least get an update on this, I was wondering about this for 2 months already :)
People ask about it on the forum too and it's a lot of confusion.

me-no-dev, 2017-02-10:
and in those 2 months everything about BT in ESP32 changed :) imagine how that would have worked

WALLTECH, 2017-02-10:
Please can someone explain how to implement IDF from the IDE? It doesn't compile plain example files from the IDF

birkir, 2017-02-27:
We now have a sample bluetooth library that works 👍 

https://github.com/espressif/arduino-esp32/blob/master/libraries/SimpleBLE/src/SimpleBLE.cpp

torntrousers, 2017-02-27:
And over here - http://developer.ibm.com/recipes/tutorials/experiments-with-bluetooth-and-watson

github4f, 2017-03-07:
@birkir Does it support ECDH or any other secure BLE connection?

ESAlexhi, 2017-05-13:
I tested SimpleBLE   and I have error:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:2232
load:0x40078000,len:11744
load:0x40080000,len:252
entry 0x40080034
ESP32 SDK: v2.0-rc1-761-g65acd99c
BTDM CONTROLLER VERSION: 010101
btip start
copy .data from 4000d890 to 3ffae6e0, len 00001830
set .bss 0x0 from 3ffb8000 to 3ffbff70, len 00007f70
BTDM ROM VERSION 0101
BD_ADDR: 24:0A:C4:03:A8:1A
NVDS MAGIC FAILED
RF Init OK with coex
Enable Classic BT
Enable Low Energy

What could be?

me-no-dev, 2017-05-13:
I do not see an error

ESAlexhi, 2017-05-13:
I do not see  
Serial.println("Press the button to change the device name");  

and  
BLE32 at: ....    :(    in my smartphone  (BLE)

ble.begin("ESP32 SimpleBLE");   //crash

jarda195, 2017-05-13:
I have the same problem...

jarda195, 2017-05-13:
Still waiting for a line 

ble.begin("ESP32 SimpleBLE");

peterrautek, 2017-05-16:
I see the same behaviour:
ble.begin("ESP32 SimpleBLE"); //crash

joeybab3, 2017-05-23:
So it's hanging on the ble.begin line as in it just freezes there?

me-no-dev, 2017-05-23:
fixed in the latest code (from yesterday)

selimmeric, 2017-05-31:
could you share any working example ..

whiskyplausible, 2017-06-01:
Yes, am also wondering if there is some example code - I'd just like to exchange data over a BLE serial link with a phone. From what I can see the library can set the name but nothing else so far - is that correct?

me-no-dev, 2017-06-01:
I think you guys misunderstand "I'd just like to exchange data over a BLE serial" with something simple :) This in particular is called SPP and is not even implemented on a base level yet.
You need to understand that the ESP32 has a bluetooth radio, that does not make it automatically able to do anything. All features that you want need to be first implemented by software and then abstracted to you so you can call BTSerial.begin() for example

whiskyplausible, 2017-06-01:
Ah I see! Yes that makes sense. So for time being perhaps best solution for me is to just use an external bluetooth module with UART I guess. 

Do you think a BTSerial.begin() will be implemented one day for the Arduino ESP32? Or perhaps is a maybe/maybe not thing?

me-no-dev, 2017-06-01:
It will surely be implemented once it's implemented on a lower level in IDF.

eraguzin, 2017-06-07:
Hi me-no-dev, since you seem to be the most knowledgeable on Bluetooth through the Arduino core, is there any Bluetooth communication that has been implemented at all yet through Arduino?

If not, how can I get started with the effort, I'm interested in making it work.

igrr, 2017-06-07:
@eraguzin Please take a look at SimpleBLE: https://github.com/espressif/arduino-esp32/tree/master/libraries/SimpleBLE/src

If you are interested in working on BLE support, that's great! I will open a new issue for discussion of approaches which can be taken.

eraguzin, 2017-06-07:
I had tried out SimpleBLE, but it looks like it only advertises as of now right? I don't see any ability to read or write data through BT.  How do I get caught up on the effort so far? Is there any documentation on how everything is organized?  I'm not familiar with the innards of the ESP32.  Thanks!

igrr, 2017-06-07:
That is correct. To read or write data you would need to work with GATT protocol either as server (peripheral) or client (central).
I have opened a meta issue for discussion of BLE support here: https://github.com/espressif/arduino-esp32/issues/423. It includes some links to get started.

techboycr, 2017-07-14:
So far no one has manage to make it work in arduino??

Would be suitable to put some IDF Like code into arduino and compile it?

RiRomain, 2017-11-07:
Just in case someone end up on this issue after a google search, some example of scan can be found here:
https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/examples/BLE_scan/BLE_scan.ino