espressif/arduino-esp32 #201: Fresh Install - esptool.py fails to connect

hoopsurfer, 2017-02-12:
I am seeing "A fatal error occurred: Failed to connect to ESP32 ..." after a fresh install.   It appears the upload is not using esptool.exe which I expected based on the output:

esptool.py v2.0-dev
connecting...

A fatal error occurred: Failed to connect to ESP32: Timed outwaiting for packet header

After cloning fresh and running get.exe I see another .zip being downloaded not in the instructions.  Any hints what might be broken.  I have an older installation that doesn't have the extra file downloaded and is working.

me-no-dev, 2017-02-12:
that extra file builds the partition scheme and has nothing to do with the problem you are having, which seems to be that the ESP can not reboot into download to properly upload.
Try holding the BOOT button (gpio0 low) while uploading.

hoopsurfer, 2017-02-13:
With the same esp32 dev board it works fine on one machine where the command is shown in the output:

python.exe C:\Users\admin.home\Documents\Arduino\hardware\espressif\esp32/tools/esptool.py --chip esp32 --port COM1 --baud 921600 write_flash -z --flash_freq 80m --flash_mode dio --flash_size 4MB 0x1000  .. etc. 

The newer installation where it fails there is no command shown  -- not sure why that is different.  What causes the esptool command echo?   I recloned and reran get on the old install and it continues to work.

hoopsurfer, 2017-02-13:
The output difference was preferences verbose output from upload checkbox.  So the output that works is running:
python.exe C:\Users\...\Documents\Arduino\hardware\espressif\esp32/tools/esptool.py ...

The installation that fails is running:
C:\Users\...\Documents\Arduino\hardware\espressif\esp32/tools/esptool.exe ...

The new installation running the new esptool.exe is failing, not clear why there is a difference, board?

me-no-dev, 2017-02-19:
esptool.exe is newer version of esptool. I have seen other reports for intermittent issues and I have seen some, but have never had issues uploading holding the BOOT button (GPIO0 LOW) while esptool is trying to start the upload.

me-no-dev, 2017-03-10:
Closing because of no response

closed 2017-03-10 by me-no-dev