espressif/arduino-esp32 #207: Application crashed with core panic

Thosch42, 2017-02-13:
My current Application crashed with the last version of arduino-esp32. Last Friday, everything worked very well. I cloned the last version this morning, started get.exe etc. everything looks good. The Application compiles fine & upload finishes with no errors, but immediately after upload:

ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:1760
load:0x40078000,len:6668
load:0x40080000,len:252
entry 0x40080034
E (1122) wifi: wifi_init 1143 ret=4363

Connecting to SmartCubeAP
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:721 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x4008343f
Guru Meditation Error: Core  1 panic'ed (abort)

Backtrace: 0x40008155:0x3ffcf510 0x40007d16:0x3ffcf530 0x40097144:0x3ffcf570 0x400f3a5c:0x3ffcf5c0 0x400ed50f:0x3ffcf5e0 0x400d14fe:0x3ffcf610 0x400d155a:0x3ffcf640 0x400d15d3:0x3ffcf660 0x400d0ed7:0x3ffcf760 0x400d0f37:0x3ffcf790 0x40105a3a:0x3ffcf7b0

CPU halted.

ExceptionDecoder give me this:

Decoding 11 results
0x40097144: pp_post at ?? line ?
0x400f3a5c: ieee80211_ioctl at ?? line ?
0x400ed50f: esp_wifi_start at ?? line ?
0x400d14fe: WiFiGenericClass::getMode() at c:\arduino\arduino-1.8.1\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/stl_vector.h line 923
:  (inlined by) WiFiGenericClass::getMode() at C:\Arduino\arduino-1.8.1\hardware\espressif\esp32\libraries\WiFi\src/WiFiGeneric.cpp line 259
0x400d155a: WiFiGenericClass::enableSTA(bool) at c:\arduino\arduino-1.8.1\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/stl_vector.h line 923
0x400d15d3: WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at C:\Arduino\arduino-1.8.1\hardware\espressif\esp32\libraries\WiFi\src/WiFiSTA.cpp line 557
0x400d0ed7: setup_wifi() at C:\Users\ts\OneDrive\!P\GCube\FW\MicroCube\TVOC2MQTT_1\TVOC2MQTT_1/TVOC2MQTT_1.ino line 81
0x400d0f37: setup at C:\Users\ts\OneDrive\!P\GCube\FW\MicroCube\TVOC2MQTT_1\TVOC2MQTT_1/TVOC2MQTT_1.ino line 62
0x40105a3a: loopTask(void*) at C:\Arduino\arduino-1.8.1\hardware\espressif\esp32\cores\esp32/main.cpp line 11 (discriminator 1)

copercini, 2017-02-13:
Same here. 

For me this issue just appear after some normal use. So I revert a lot of commits and the problem continues. The ESP just back to work after flash a blank bin. Now it's working fine with last version of this git and I not sure where is the bug.

hcs-svn, 2017-02-13:
Just got the same problem.
Suddenly it stopped working. As soon as I called a WiFi method (does not matter which one) it crashed.
`esptool --port COM3 erase_flash`
fixed the problem.

me-no-dev, 2017-02-13:
hmm... this is rather strange, though I have a clue where is the problem, I'm not exactly sure why it happens. I slightly changed the partition scheme which made NVS smaller with 8K and some data does stay there, but at the same time I made that change when I enabled WiFi data to be stored in NVS, so in theory, unless you had run at some point with setting to NVS, you should have been fine.

I'm thinking of spending some time soon and making an IDE plugin to easily wipe, reflash and so on.

RodgerGr, 2017-02-17:
Same for me.
The command from @hcs-svn  did the trick and everything is well again.

igrr, 2017-03-14:
We happened to figure out the root cause of this issue today.

In some cases, when the board was first used with the default non-OTA partition table, and was later used with OTA-enabled partition table, NVS partition may be corrupted after some time (or immediately after such an update).

In the non-OTA partition table, NVS was allocated 6 sectors of space. But in the OTA-enabled partition table, NVS is allocated only 4 sectors. This means that when partition table is changed, NVS library will only "see" the first 4 sectors of the old 6-sector partition.

One of the invariants maintained by the NVS is that it always has at least one spare sector which is not used to store data, but is used when moving things around. When NVS partition is cropped from 6 sectors to 4 sector, it is possible that the only spare sector is no longer part of NVS partition. NVS didn't check this invariant upon initialization, and would only error out with `ESP_ERR_NVS_INVALID_STATE` error when it would check this invariant during one of `nvs_set_blob` calls from wifi initialization process. This was propagated out of `esp_wifi_init` and logged as `E (1122) wifi: wifi_init 1143 ret=4363`.

We will update NVS to detect and handle the case of cropped partitions better, when possible, and we will update NVS documentation with the instructions about dealing with those issues. In the meantime, consider erasing flash using `make erase_flash` (if using ESP-IDF) or using esptool  as described above (if using Arduino).

gellis12, 2017-04-23:
I'm having exactly the same issue, but using esptool to erase the flash didn't work. Anyone have any other ideas?

kyjak, 2017-06-24:
Same issue here, erasing flash did not help. Using arduino 1.8.3 with espressif.

bill-orange, 2017-09-26:
Ditto.  When using the example's code, I get a core panic.

me-no-dev, 2017-09-26:
@bill-orange check your USB cable. Make sure power is stable too. There are no known issues to prevent you from running example sketch

bill-orange, 2017-09-26:
I should have done more testing before my post. 

I retested and the OTA example ran fine, if unmodified. The failure occurred in my sketch when I added the example's code to it.   To pin down the problem, I added my  #include's into the working example line by line.  The inclusion of  #include "esp_system.h"  into the OTA example  causes the crash. Can someone confirm?

me-no-dev, 2017-09-26:
I do not see how ```#include "esp_system.h"``` would cause issues ;) what are you using actually from that header

beegee-tokyo, 2017-09-26:
I included **`esp_system.h`** and use some functions (_`esp_chip_info(&espInfo);`, `esp_read_mac(baseMac, ESP_MAC_WIFI_STA);`_) and have no problems. No crash, no problem with OTA.

bill-orange, 2017-09-26:
I don't see why one should affect the other either.  I am using esp_restart(), however for the purposes of this test I merely included #include "esp_system.h" to the build.  No other line of code was added.

mrengineer7777, 2023-03-29:
Closing stale issue

closed 2023-03-29 by mrengineer7777