espressif/arduino-esp32 #208: I2C does not transmit correct signal

yukini3, 2017-02-15:
Hi, the following code demonstrates sending two packs of I2C data to I2C EEPROM. When I view the SDA line with the scope, I only can see the first pack of I2C data is transmitted but not 2nd one. 2nd packet seems only one byte. Inserting a delay in between sending packets (I did delayMicrosecods(1300)) could send 2nd packet correctly.

#include <Arduino.h>
#include <Wire.h>

#define PIN_SDA		21			// I2C SDA
#define PIN_SCL		22			// I2C SCL

void setup()
{
	int i2cdevice = 0x57;		// AT24C32 address
	int addr;
	int i;

	Serial.begin(115200);
	Serial.println("Initializing I2C");
	Wire.begin(PIN_SDA, PIN_SCL);	

	addr = 0;

	Wire.beginTransmission(i2cdevice);
	Wire.write(addr >> 8);
	Wire.write(addr & 0xff);
	for (i = 0; i < 8; i++) {
		Wire.write(i);
	}
	Wire.endTransmission();

	addr += 8;

	//delayMicroseconds(1300);

	Wire.beginTransmission(i2cdevice);
	Wire.write(addr >> 8);
	Wire.write(addr & 0xff);
	for (i = 8; i < 16; i++) {
		Wire.write(i);
	}
	Wire.endTransmission();
}

void loop()
{
	Serial.print("hello");
	delay(9000);
}

This is the screenshot of I2C signal.

[image]

me-no-dev, 2017-02-22:
sorry to respond so late :) I'll look for eeprom like that, I think I have a few somewhere and give it a go :) I was hoping to not see another I2C device fail.

yukini3, 2017-02-23:
Hi, thanks for taking care of this. I tested with the SDK version v2.0-rc1-118-gbf20a0a and still the same.

arcao, 2017-03-03:
The problem is that EEPROM needs some time to finish physical writing of data. This also mentioned in a [datasheet](http://www.atmel.com/images/doc0336.pdf):

> ... Following receipt of the 8-bit data word, the EEPROM will output a zero and the addressing device, such as a microcontroller, must terminate the write sequence with a stop condition. At this time the EEPROM enters an internally-timed write cycle, t<sub>WR</sub>, to the nonvolatile memory. All inputs are disabled during this write cycle and the EEPROM will not respond until the write is complete (refer to Figure 2).

Solution is to use ACKNOWLEDGE POLLING:
> Once the internally-timed write cycle has started and the EEPROM inputs are disabled, acknowledge polling can be initiated. This involves sending a start condition followed by the device address word. The read/write bit is representative of the operation desired. Only if the internal write cycle has completed will the EEPROM respond with a zero, allowing the read or write sequence to continue.
```cpp
void waitEEReady(int i2cdevice) {
  #define I2C_WRITEDELAY_US  5000

  uint32_t before = micros();

  // Wait until EEPROM gives ACK again.
  // this is a bit faster than the hardcoded 5 milli
  while ((micros() - before) <= I2C_WRITEDELAY_US) {
    Wire.beginTransmission(i2cdevice);
    int x = Wire.endTransmission();
    if (x == 0) break;
  }

  Serial.printf("Delay: %u\n", (micros() - before));
}
```

By my tests 24C256 EEPROM needs around 3440 us to write data.

yukini3, 2017-03-03:
Thank you for your detailed description about the problem. I now see the problem is in my code. I apologize that I thought that this is a problem of I2C drivers.

closed 2017-03-03 by yukini3