espressif/arduino-esp32 #209: Git Gui clone repository - virus Gen:Variant.Graftor.340977

sveinutne, 2017-02-16:
WhenItry to clone from:
Source Location: https://github.com/espressif/arduino-esp32.git
deleted file mode 100644
Binary files a/tools/gen_esp32part.exe and /dev/null differ

get.exe and gen_esp32part.exe is removed by thevirus protection program.
Gen:Variant.Graftor.340977

me-no-dev, 2017-02-16:
tell you antivirus not to scan the folder. Source of the files is there as well ;) in case you worry something might be up. Feel free to install python and regenerate them also :)

sveinutne, 2017-02-16:
Removed F-secure, and used Windows Defender. No virus wornings, so I guess it will be OK. Thanks

Closed #209<https://github.com/espressif/arduino-esp32/issues/209>.

-