espressif/arduino-esp32 #212: PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO16_U, FUNC_GPIO16_EMAC_CLK_OUT) problem

ESP32DE, 2017-02-16:
https://github.com/espressif/esp-idf/issues/345

hi 
how is the stand in arduino-esp32?

best wishes
rudi ;-)

me-no-dev, 2017-02-16:
how is this connected to Arduino in any way :) it's register manipulations that are the same here and in idf. Play with the registers and you will get it going.

ESP32DE, 2017-02-16:
example if arduino user want extern clk for phy set and enable?
is there a help or description available?

as an outsider and without RMII / Register document is the play with registers not effective in development time, sry

we need to set clk, how
we need to set remap? 

perhabs i will play - perhabs not, 
if not, then the theme is closed to me for future to ask for it and silence comes for me long time too. 

it makes not fun, to ask things, that is not described, and play with things, that perhabs not run in V0 or run. so "much playtime" i have not  - sade for the restrained support in this theme.

best wishes
rudi ;-)

closed 2017-02-16 by ESP32DE

me-no-dev, 2017-02-16:
I'm not sure I understand everything you wrote, but I get the feeling you expect this to be some sort of different knowledge base than IDF when this is build on top of IDF and the knowledge and information actually comes from there first. As for the clock out, I'll worry about it when I start Ethernet integration in Arduino, but by then IDF will probably have the documentation and everything else sorted.

ESP32DE, 2017-02-16:
I wonder how I can build a chip without first knowing how to set which registers.

There must be a certain knowledge management, so that the team can work together on the IDF / Arduino32.
I do not have the opportunity to go to the office right next door
 The Espressif team to ask state of affairs to a problem,
To continue the project.

I wonder if several people in the forum ask for the possibility why none of espressif team gives an effective answer, how to set the register then. There is a silence about it.

There are several questions to IDF in the forum on this topic GPIO16 as CLK to use, and we get no answer.

Https://esp32.com/viewtopic.php?f=12&t=1119&start=10
Https://esp32.com/viewtopic.php?f=12&t=367
Https://esp32.com/viewtopic.php?f=2&t=950

CAN was the same theme,
Jeroen gave a state

Https://esp32.com/viewtopic.php?f=12&t=380

And we knew how to get to the register,
And the CAN was born.

I do not bet for code, I bettle the procedure at the registers.
There is no docu about it at the moment.
But there is an ethernet demo available.

So knowledge must be there.

This uses an extrene CLK for the PHY.

As you know, Olimex gave me the ESP32-EVB and on this the GPIO16 is planned as CLK since GPIO16 was so described in the technical reference that it works.

Which registers and how - this is unfortunately nowhere.

"Play with the registers and you get it"
I'm sorry for not Christo, sorry about the time.

I'll pause for a long time - I think I've been faithful enough to you for a long time, but it makes little sense to be so dependent on IoT.

You're sitting at the source - I do not.

Once you get the necessary information and again is long silence in the forums to simple things.

It does not carry any further.
it's OK.

Is nothing personal or against you.
It just jammed up again, needed information to run after.
Regardless of whether IDF, Forum or Uno Support.

best wishes
rudi ;-)

LyraT and LyraP the same theme.
since Nov 16 , there are audio, PLL things available
but we get not the Info we need christo,
thats the problem. but is ok.

me-no-dev, 2017-02-16:
rudi you are flipping again with no reason ;) I am not sitting in the office, I'm thousands of kilometers away and do not have the access to many many things. I can also tell you that hardware is not designed in the way that it's in your head. People that design hardware and people that implement it are two different kinds of people. Until the hardware team gives proper specs, the software people have no way of knowing, except "Play with the registers and you get it" hopefully.
And i'm like levels away from the software people that have access to those hardware people, so sorry but I feel offended now.

me-no-dev, 2017-02-16:
and btw my name is Hristo

ESP32DE, 2017-02-16:
sry for typo, Christo / Hristo,
somy guys write rudy / rudi..
..

You do not need to feel offended because there is no reason
I said nothing personal or against you.

If I had "hurt" you, then I would apologize publicly, that was not my intention.

I feel myself left alone,

Anybody I appeal does not feel addressed
But everyone works for espressif in software development.
That's what I do not understand.

I will be in the future now with questions and keep espressif directly contact as I did it at the beginning.
I wait for 3 weeks and more but somehow always see something through.

Head up, I said-nothing personal or against you,
But I can not start with the answer, we are only playing around here,
Because the docus / infos are missing.

OK?
are we friends again?
hope so

best wishes
rudi ;-)